libts/tslib #126: ts_uinput does not produce any  output

bmoraine, 2018-03-26:
Hello,

After succesfully used tslib 1.15 to manage my resistive touchscreen under QT5.5 with QT tslib plugin, I try to use ts_uinput in order QT deals with my touchscreen at EVDEV level to support Touchscreen rotation.

When I launch ts_uinput -d -v, a new /dev/input/eventX is mounted but when I test it there is no output.

For your information, 
- I used default /etc/ts.conf file with only module_raw input and module linear uncommented
- pointercal file content is : "5755 29 -1828536 187 -8683 33009782 65536 320 480"

# export TSLIB_TSDEVICE=/dev/input/event0
# export TSLIB_CALIBFILE="/mnt/kap_data/sys/pointercal"
# ts_uinput -v -d &> /dev/kmsg
[   55.390000] input: ts_uinput as /devices/virtual/input/input1
[   55.410000] ts_uinput: See the kernel log for the device number
# ls /dev/input
event0  event1
# evtest /dev/input/event0
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "at91_adc"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max     4095
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max     4095
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max   16777215
Properties:
Testing ... (interrupt to exit)
Event: time 1522076410.696493, type 3 (EV_ABS), code 0 (ABS_X), value 2077
Event: time 1522076410.696493, type 3 (EV_ABS), code 1 (ABS_Y), value 1746
Event: time 1522076410.696493, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 8927
Event: time 1522076410.696493, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1522076410.696493, -------------- SYN_REPORT ------------
Event: time 1522076410.712496, type 3 (EV_ABS), code 0 (ABS_X), value 2054
Event: time 1522076410.712496, type 3 (EV_ABS), code 1 (ABS_Y), value 1776
Event: time 1522076410.712496, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 6438
Event: time 1522076410.712496, -------------- SYN_REPORT ------------
Event: time 1522076410.728496, type 3 (EV_ABS), code 0 (ABS_X), value 2035
Event: time 1522076410.728496, type 3 (EV_ABS), code 1 (ABS_Y), value 1774
Event: time 1522076410.728496, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 4842
Event: time 1522076410.728496, -------------- SYN_REPORT ------------
Event: time 1522076410.744494, type 3 (EV_ABS), code 0 (ABS_X), value 2030
Event: time 1522076410.744494, type 3 (EV_ABS), code 1 (ABS_Y), value 1806
Event: time 1522076410.744494, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 3910
# evtest /dev/input/event1
Input driver version is 1.0.1
Input device ID: bus 0x6 vendor 0x0 product 0x0 version 0x0
Input device name: "ts_uinput"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max      319
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max      479
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max   16777215
Properties:
Testing ... (interrupt to exit)
[ Nothing occurs when I use my TS]

I maybe miss something in ts_uinput configuration, could you help me?

Regards

Baptiste

merge, 2018-03-27:
I'm not sure yet, but we definitely should support your case. Could you record something (short) from your original device (event0) to a file using evemu-record and upload the file here? thanks!

bmoraine, 2018-03-27:
Hello,

First thank you for your support!

I executed  then following commands :

#export TSLIB_TSDEVICE=/dev/input/event0
# export TSLIB_CALIBFILE="/mnt/kap_data/sys/pointercal"
# ts_uinput -v -d &> /dev/kmsg
[   52.410000] input: ts_uinput as /devices/virtual/input/input1
[   52.430000] ts_uinput: See the kernel log for the device number
# ls /dev/input/
event0  event1

then executed evemu-record on event0

Please find attached evemu-record output : [
[event0.log](https://github.com/kergoth/tslib/files/1851187/event0.log)
](url)

For information, please find the output of ts_uinput when not daemonized
# ts_uinput -v 

tslib environment variables:
       TSLIB_TSDEVICE: '/dev/input/event0'
      TSLIB_PLUGINDIR: '(null)'
  TSLIB_CONSOLEDEVICE: '(null)'
       TSLIB_FBDEVICE: '(null)'
      TSLIB_CALIBFILE: '/mnt/kap_data/sys/pointercal'
       TSLIB_CONFFILE: '(null)'

ts_uinput: using framebuffer device /dev/fb0
ts_uinput: using input device /dev/input/event0
[  616.230000] input: ts_uinput as /devices/virtual/input/input2
ts_uinput: We don't use a multitouch device
ts_uinput: running uinput version 3
ts_uinput: See the kernel log for the device number

Please note I use a 3.10 kernel, is there any chance that the uinput driver version is too old to be compatible with ts_uinput?

Regards

Baptiste

merge, 2018-03-27:
Thanks! So I run your sample, use it as input for ts_uinput (-i) and do evemu-record ts_uinput's new device now:

[event0_uinput_evemu_record.txt](https://github.com/kergoth/tslib/files/1851460/event0_uinput_evemu_record.txt)

you have to ignore the actual x/y values as I don't care about my ts.conf setup here, but it works as it's pressing / moving / releasing.

I'm no aware of critical changes, but indeed i've never tried such an old kernel (and it's uinput version 3). Can't you run a recent kernel?

As a sidenote: You shouldn't need to set TSLIB_TSDEVICE explicitely, but it doesn't matter.

bmoraine, 2018-03-27:
Hello,

Thank you for your feedback!

I have quickly backported uinput driver version 0.4 that has been introduced in kernel 3.15 and now ts_uinput produces some output ans so has a normal behaviour.

https://elixir.bootlin.com/linux/v3.15/source/drivers/input/misc/uinput.c

I am now able to use in QT5.5 my resistive touchscreen at EVDEV level with rotation capabilities.

I was wondering if there is a way to do the same without using ts_uinput for example using a dynamically generated pointercal file derivated from original pointercal file according desired rotation 

Regards   

Baptiste

merge, 2018-03-27:
ok. We maybe should fail in case of uinput version < 4... I guess we could do so as it "only" affects the ts_uinput application.

The "pointercal" file is read by tslib's "linear" filter. Honestly, I don't know if there's any evdev driver that reads such (or a similar) file.

A more straight forward way, if you're into C++ would be to update Qt's tslib platform system module to support multitouch, if the version of libts provides it (TSLIB_VERSION_MT is defined).

bmoraine, 2018-03-27:
Hello,

I am using tslib 1.15 and support of multitouch is enabled.

I am not sure to understand your last comment.

I had a quick look to tslib QT management 
https://github.com/qt/qtbase/blob/5.11/src/platformsupport/input/tslib/qtslib.cpp
and did not see any evolution for multitouch support.

Regards

Baptiste

merge, 2018-03-27:
I meant, yes, there is no evolution for multitouch support in Qt but you could add it if you need it and like programming in C++. That'd be a way for you if *don't* want to use ts_uinput.

bmoraine, 2018-03-27:
ts_uiput is fine.  

Thank you very much for your support

Regards

commit 932bb4f2 (2018-03-28): ts_uinput: fail for unsupported old kernel versions

According to tests we need the kernel's uinput version >= 4. Since
it got introduced in Linux v3.15 with is very old, we simply fail
and say so in that case.

Closes #126
closed 2018-03-28 by merge