libts/tslib #127: ts_test wont close and screen wont calibrate manually

aceLeVillegas, 2018-04-14:
https://github.com/kergoth/tslib/issues/29 touched on this problem. I've tried everything to exit from the program even to ssh into the Pi to try and kill the program there and still with no luck. I think there is a need for an exit button the only buttons that appear for me is the Draw and Drag. The only way I can exit is if I unplug the Pi from the power source. Also my screen will not properly calibrate but when using the draw and drag the screen seems to be calibrated until I am on the Image for the Pitft 3.5" screen the cursor is on the opposite side of the screen.  I assume I have this problem since I cannot exit from the ts_test program and have it set up that calibration.

merge, 2018-04-15:
thanks for reporting this. unfortunately I don't have this particular device. So, how does `ts_calibrate` fail? ts_test doesn't _do_ anything. It simply shows you, how your touch-experience according to your ts.conf file feels. If you have `module linear` in there, then transformation according to your `ts_calibrate` calibration is applied.

Also, how to you install tslib? And what version of tslib are you running?

thanks so far

merge, 2018-04-15:
I assume you can connect a serial cable, log in and `killall ts_test`. But only `ts_calibrate` sets up calibration! If that doesn't run and create a calibration file (that the "linear" filter then reads) for you, we have a problem :)

but something is odd about this device or setup. I have run ts_test on 480x272 resolution screens without a problem...

aceLeVillegas, 2018-04-17:
So ts_calibrate when I set it it seems to capture where I'm tapping but when I touch the screen the mouse cursor appears mirrored to where my finger is at.  I don't understand why this happens because when I'm doing ts_test it works perfectly fine for drawing and touching. I followed this guide https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/resistive-touchscreen-manual-install-calibrate and I upgraded all the packages.

aceLeVillegas, 2018-04-17:
So when I touch the screen from left to right (Landscape) the cursor moves up and down and vice versa. On my /boot/config.txt I have:

[pi1]
device_tree=bcm2708-rpi-b-plus.dtb
[pi2]
device_tree=bcm2709-rpi-2-b.dtb
[all]
dtparam=spi=on
dtparam=i2c1=on
dtparam=i2c_arm=on
dtoverlay=pitft35-resistive,rotate=270,speed=20000000,fps=20 

the [all] was given by the adafruit-pitft-helper

merge, 2018-04-17:
To know for sure I'd need the device, but it seems like tslib works just fine. I don't know what `dtoverlay` with `rotate` does exactly, but you shouldn't need it with tslib (I think ts_test would work without it too).

It simply seems like your "real" input system doesn't take tslib's samples. It just ignores it. I'd look into our `ts_uinput` documentation (or use xf86-input-tslib. but take version 0.0.7 if you do )

aceLeVillegas, 2018-04-17:
Thank you for your help. I will look into that option.

merge, 2018-04-17:
They really seem to be missing the (important) last steps in their documentation. feel free to ask with any problems you see. The `ts_uinput` program is directly supported here and should always work.

If you want me to look into your device a bit, you can send (or attach here) an evemu-recording, using the (original) touchscreen device. `evemu-record > recording.evemu` and do a random short touch-sequence. (I can replay this and reproduce any tslib-issue in case there is any)

aceLeVillegas, 2018-04-18:
[recording.evemu.zip](https://github.com/kergoth/tslib/files/1925724/recording.evemu.zip)
I've attached the file you requested.

merge, 2018-04-19:
thanks a lot! seems like we support this device just fine. again, check what version of tslib you are running.

And make sure the `ts_test_mt` program works too, just like `ts_test` (because that's what you get when you use ts_uinput to forward touches to your evdev system config).

merge, 2018-04-19:
Also, please post how your `ts.conf` config file (most probably in /etc) looks like. Just like in #125 make it look like so:
```
module_raw input
module linear
```
nothing more.

merge, 2018-09-07:
Please feel free to open a new or reopen this issue in case you have problems still. I'm just cleaning up.

closed 2018-09-07 by merge