libts/tslib #132: when i cross compile it in ubuntu12.04,i get this error about "UI_GET_SYSNAME"

wbaiting, 2018-05-24:
undefined reference to `UI_GET_SYSNAME'

I think this may be due to my compiler version,the version of my cross compiler is 4.6

Can anyone tell me how to solve it? Thank you

merge, 2018-05-24:
yes. This is because of a bug in ts_uinput on old kernels. I recently fixed that for _running_ on old kernels, but thanks for pointing that out. It should now be fixed for building against old kernels too.

If you can, pull the master branch from git, do `autogen.sh` and configure and build it from there. It'll be in version 1.17.

merge, 2018-05-24:
of course you can instead `./configure --disable-tools`, but you won't get ts_uinput then, which you probably need...

merge, 2018-05-25:
I guess that resolves your issue. Please re-open if not. And if you want that, give your name and email address, and I'll add you to the THANKS file.

closed 2018-05-25 by merge

wbaiting, 2018-05-25:
My pleasure. My name is Bin Wang,my e-mail address is wangbinxiaoji@gmail.com. Best wishes for the project!