libts/tslib #133: ts_uinput: BTN_TOUCH no button up event when using median or variance filter

pfink-christ, 2018-06-27:
Hi,
I stumbled into the following problem and wanted to ask if this combination is not supposed to work of if it is expected to work and it is a bug. Maybe it is related or similar to #120 

Resistive touch on /dev/input/touchscreen0 via i2c.
The problem occurs when activating variance oder median filter (other filters not tested). dejitter is ok.

When using these filters I get a single BTN_TOUCH 1 event after starting evtest. The following BTN_TOUCH events are simply not reported.

~# evtest /dev/input/touchscreen0 
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "ar1021 I2C Touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   2491
      Min        0
      Max     4095
    Event code 1 (ABS_Y)
      Value   2881
      Min        0
      Max     4095
Properties:
Testing ... (interrupt to exit)
Event: time 1530093846.346035, type 3 (EV_ABS), code 0 (ABS_X), value 2779
Event: time 1530093846.346035, type 3 (EV_ABS), code 1 (ABS_Y), value 2776
Event: time 1530093846.346035, -------------- SYN_REPORT ------------
Event: time 1530093846.357528, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1530093846.357528, -------------- SYN_REPORT ------------
Event: time 1530093846.362575, type 3 (EV_ABS), code 1 (ABS_Y), value 2780
Event: time 1530093846.362575, -------------- SYN_REPORT ------------
Event: time 1530093846.367629, type 3 (EV_ABS), code 1 (ABS_Y), value 2782
Event: time 1530093846.367629, -------------- SYN_REPORT ------------
Event: time 1530093846.381629, type 3 (EV_ABS), code 0 (ABS_X), value 2780
Event: time 1530093846.381629, -------------- SYN_REPORT ------------
Event: time 1530093846.395676, type 3 (EV_ABS), code 1 (ABS_Y), value 2783
Event: time 1530093846.395676, -------------- SYN_REPORT ------------
Event: time 1530093846.409748, type 3 (EV_ABS), code 0 (ABS_X), value 2779
Event: time 1530093846.409748, -------------- SYN_REPORT ------------
Event: time 1530093846.457725, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1530093846.457725, -------------- SYN_REPORT ------------

starting ts_uinput:

TSLIB_PLUGINDIR=/usr/lib/ts TSLIB_CALIBFILE=/etc/pointercal TSLIB_CONFFILE=/etc/ts.conf TSLIB_TSDEVICE=/dev/input/touchscreen0 ts_uinput -v -s 1

tslib environment variables:
       TSLIB_TSDEVICE: '/dev/input/touchscreen0'
      TSLIB_PLUGINDIR: '/usr/lib/ts'
  TSLIB_CONSOLEDEVICE: '(null)'
       TSLIB_FBDEVICE: '(null)'
      TSLIB_CALIBFILE: '/etc/pointercal'
       TSLIB_CONFFILE: '/etc/ts.conf'

ts_uinput: using framebuffer device /dev/fb0
tslib: WARNING: no multitouch using VARIANCE filter
ts_uinput: using input device /dev/input/event1
ts_uinput: We don't use a multitouch device
ts_uinput: running uinput version 5
created /sys/devices/virtual/input/input21
/dev/input/event4
tslib: WARNING: no multitouch using VARIANCE filter

I even specified -s 1 (i.e. single touch?) Is there another possible configuration to limit tslib to single touch?

Output from event4:

~# evtest /dev/input/event4
Input driver version is 1.0.1
Input device ID: bus 0x6 vendor 0x0 product 0x0 version 0x0
Input device name: "ts_uinput"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    782
      Min        0
      Max     1279
    Event code 1 (ABS_Y)
      Value    565
      Min        0
      Max      799
Properties:
Testing ... (interrupt to exit)
Event: time 1530093846.346218, type 3 (EV_ABS), code 0 (ABS_X), value 875
Event: time 1530093846.346218, type 3 (EV_ABS), code 1 (ABS_Y), value 543
Event: time 1530093846.346218, -------------- SYN_REPORT ------------
Event: time 1530093846.367780, type 3 (EV_ABS), code 1 (ABS_Y), value 544
Event: time 1530093846.367780, -------------- SYN_REPORT ------------
Event: time 1530093846.409884, type 3 (EV_ABS), code 1 (ABS_Y), value 545
Event: time 1530093846.409884, -------------- SYN_REPORT ------------

I tried the median filter instead as suggested in the Readme, but this gives me similar results.

Is this configuration supposed to work with the default ts.conf?

If I remove the variance filter I get all necessary Events.

Best Regards,
Peter

merge, 2018-06-27:
thanks for reporting. I'm somewhat confused about your various ts.conf settings. Would you paste the working and not-working files here for reference? Generally, "variance" should be considered deprecated as of now. It causes more trouble than it helps.

pfink-christ, 2018-06-27:
the default (through yocto & tslib-1.1.5) looks like this:

```
# Uncomment if you wish to use the linux input layer event interface                                                                                                                                                
module_raw input                                                                                                                                                                                                    
                                                                                                                                                                                                                    
# Uncomment if you're using a Sharp Zaurus SL-5500/SL-5000d                                                                                                                                                         
# module_raw collie                                                                                                                                                                                                 
                                                                                                                                                                                                                    
# Uncomment if you're using a Sharp Zaurus SL-C700/C750/C760/C860                                                                                                                                                   
# module_raw corgi                                                                                                                                                                                                  
                                                                                                                                                                                                                    
# Uncomment if you're using a device with a UCB1200/1300/1400 TS interface                                                                                                                                          
# module_raw ucb1x00                                                                                                                                                                                                
                                                                                                                                                                                                                    
# Uncomment if you're using an HP iPaq h3600 or similar                                                                                                                                                             
# module_raw h3600                                                                                                                                                                                                  
                                                                                                                                                                                                                    
# Uncomment if you're using a Hitachi Webpad                                                                                                                                                                        
# module_raw mk712                                                                                                                                                                                                  
                                                                                                                                                                                                                    
# Uncomment if you're using an IBM Arctic II                                                                                                                                                                        
# module_raw arctic2                                                                                                                                                                                                
                                                                                                                                                                                                                    
module pthres pmin=1                                                                                                                                                                                               
module variance delta=30                                                                                                                                                                                                                                                                                                                                                                                    
module dejitter delta=100                                                                                                                                                                                           
module linear
```

the following is working:
[... cut ...]
module pthres pmin=1                                                                                                                                                                                                                                                                                                                                                                                       
module dejitter delta=100                                                                                                                                                                                           
module linear
```

pfink-christ, 2018-06-27:
good to know that it's deprecated.

this did not work either:
```
module_raw input
module pthres pmin=1                                                                                                                                                                                               
#module median depth=5
module dejitter delta=100
module linear
```