libts/tslib #134: Configuring touch device on Windows

skirk, 2018-07-28:
Hi, I've compiled the 1.16 library using cygwin and made a client app based on the example from the documentation. I have some runtime crash problems in ts_setup. 

I have also tried the 1.11 Windows testing binaries from the documentation but the executables end without any print out.

Should windows automatically detect devices or is there some Windows specific configuration that needs to be done and where should the config files exist?

merge, 2018-07-29:
ts_setup() will always fail on Windows. There's no module_raw library module for reading input on Windows. You'd need to write that first.

Am 28. Juli 2018 09:29:58 MESZ schrieb skirk <notifications@github.com>:
> Hi, I've compiled the 1.16 library using cygwin and made a client app
> based on the example from the documentation. I have some runtime crash
> problems in ts_setup.
>
> I have also tried the 1.11 [...]

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

skirk, 2018-08-06:
I'd interested in contributing such module. 

This is slightly off-topic of this issue but in order to dev/test on windows I would like to have a sane way to build/test on Visual Studio. For this I'd be eager to add a CMake based build system. Is this something you would be interested in adding to the project or would you rather only have the autotools/Make option?

merge, 2018-11-21:
Are you still working on a windows module? If not, maybe you could summarize why not. thanks!

skirk, 2018-12-11:
Hi @merge , 
I'm not actively working on it at the moment. I have some experimental version somewhat working but is by no means production ready. 

The main issue is the device model is entirely different. On unix-like systems you can grab the events from low level fd so encapsulation is quite easy as you can just do fopen on the file descriptor. On Windows and similar higher level touch implementations (X, Wayland), the touches come via the applications window. So in order for tslib to process on those events it would need to have a handle to the apps window. It can be done by repurposing for example the device fd to that handle but it is a lot of #ifdefs and generally very hacky as the client app would also need to call some special hook functions related to window input to make any of it work. 

To properly support windowed touches would be a drastic design change and perhaps not in the scope of this library.

merge, 2018-12-12:
hm. I can only think into the blue, but you'd probably have to think of a tslib windows plugin as a touchscreen device driver for windows. except that we read existing input and provide our API. [This](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/keyboard-and-mouse-hid-client-drivers) seems to describe what we need at a high level.

Of course then nobody yet uses your API, but that'd be a different task, maybe going in [this](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/virtual-hid-framework--vhf-) and [this](https://stackoverflow.com/questions/47942366/windows-virtual-mouse-driver) direction).