libts/tslib #135: Xorg startup segfault with Waveshare USB TS

MWP, 2018-08-06:
Greetings,

I'm trying to get my rPi up and running. It uses a Waveshare USB capacitive touch screen.

dmesg shows:
```
[    2.879082] usb 1-1.4: Product: WaveShare Touchscreen
[    2.886232] usb 1-1.4: Manufacturer: WaveShare
[    2.903051] input: WaveShare WaveShare Touchscreen as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:0EEF:0005.0001/input/input0
[    2.920970] hid-generic 0003:0EEF:0005.0001: input,hidraw0: USB HID v1.11 Device [WaveShare WaveShare Touchscreen] on usb-3f980000.usb-1.4/input0
[    2.982676] input: WaveShare WaveShare Touchscreen as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:0EEF:0005.0001/input/input1
[    3.001615] hid-multitouch 0003:0EEF:0005.0001: input,hidraw0: USB HID v1.11 Device [WaveShare WaveShare Touchscreen] on usb-3f980000.usb-1.4/input0
```

On attempting to startx i get a sigfault error.
The Xorg log shows:
```
[  7456.282] (II) config/udev: Adding input device WaveShare WaveShare Touchscreen (/dev/input/mouse0)
[  7456.282] (**) WaveShare WaveShare Touchscreen: Applying InputClass "tslib touchscreen catchall"
[  7456.282] (**) WaveShare WaveShare Touchscreen: Applying InputClass "tslib catchall"
[  7456.282] (II) LoadModule: "tslib"
[  7456.283] (II) Loading /usr/lib/xorg/modules/input/tslib_drv.so
[  7456.283] (II) Module tslib: vendor="X.Org Foundation"
[  7456.283]    compiled for 1.20.0, module version = 1.1.1
[  7456.283]    Module class: X.Org XInput Driver
[  7456.283]    ABI class: X.Org XInput driver, version 24.1
[  7456.283] (II) Using input driver 'tslib' for 'WaveShare WaveShare Touchscreen'
[  7456.283] (**) WaveShare WaveShare Touchscreen: always reports core events
[  7456.283] (**) Option "path" "/dev/input/mouse0"
[  7456.342] (II) tslib: WaveShare WaveShare Touchscreen: using libts version 901
[  7456.342] (**) WaveShare WaveShare Touchscreen: always reports core events
[  7456.342] (EE) tslib: WaveShare WaveShare Touchscreen: ioctl EVIOCGBIT failed(EE) PreInit returned 2 for "WaveShare WaveShare Touchscreen"
[  7456.342] (EE) Segmentation fault at address 0x0
```
If I move or rename /usr/lib/xorg/modules/input/tslib_drv.so so it's not loaded, X starts up ok.
(no working touchscreen though obviously).

Any ideas?

Thanks!

merge, 2018-08-06:
xf86-input-tslib 1.x.x is broken. Use 0.0.7. until it's fixed. This is no tslib issue, it's xf86-input-tslib specific. Alternatively, you don't use xf86-input-tslib at all, and use xf86-input-evdev and ts_uinput (which is included in tslib tools), see the tslib README documentation.

MWP, 2018-08-06:
Thankyou, im now using 0.0.7.
That is that issue solved.

Now I get this repeated very quickly on the console:
```
tslib: Selected device is not a Linux input event device
tslib: Selected device is not a Linux input event device
tslib: Selected device is not a Linux input event device
XIO:  fatal IO error tslib: Selected device is not a Linux input event device
0 (tslib: Selected device is not a Linux input event device
Successtslib: Selected device is not a Linux input event device
) on X server ":0tslib: Selected device is not a Linux input event device
"
tslib: Selected device is not a Linux input event device
      after 520tslib: Selected device is not a Linux input event device
 requests (520tslib: Selected device is not a Linux input event device
 known processed) with 0tslib: Selected device is not a Linux input event device
 events remaining.
tslib: Selected device is not a Linux input event device
tslib: Selected device is not a Linux input event device
tslib: Selected device is not a Linux input event device
tslib: Selected device is not a Linux input event device
```

merge, 2018-08-06:
and what version of tslib are you running here?

MWP, 2018-08-06:
It's 1.1.6.

merge, 2018-08-07:
alright then. could you post the output of evtest for your device?

MWP, 2018-08-08:
Is this what you meant? I'm not sure if this helps?

When I start X, I get no errors until I tap on the touchscreen, which is when i get the flood of "tslib: Selected device is not a Linux input event device" messages.

In case it makes any difference, this is a raspberry Pi CM3, running a OS built by buildroot.

```
root@rpidash:/etc/X11# lsusb
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 002: ID 1a40:0101
Bus 001 Device 004: ID 0eef:0005
Bus 001 Device 003: ID 0b95:7e2b
root@rpidash:/etc/X11# dmesg | grep input
[    2.378643] udevd[91]: specified group 'input' unknown
[    2.911484] input: WaveShare WaveShare Touchscreen as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:0EEF:0005.0001/input/input0
[    2.928824] hid-generic 0003:0EEF:0005.0001: input,hidraw0: USB HID v1.11 Device [WaveShare WaveShare Touchscreen] on usb-3f980000.usb-1.4/input0
[    2.977062] input: WaveShare WaveShare Touchscreen as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:0EEF:0005.0001/input/input1
[    2.994907] hid-multitouch 0003:0EEF:0005.0001: input,hidraw0: USB HID v1.11 Device [WaveShare WaveShare Touchscreen] on usb-3f980000.usb-1.4/input0
root@rpidash:/etc/X11# ll /dev/input/
     0 drwxr-xr-x    2 root     root          80 Jan  1 00:00 .
     0 drwxr-xr-x   10 root     root        2.9K Jan  1 00:00 ..
     0 crw-------    1 root     root       13,  63 Jan  1 00:00 mice
     0 crw-------    1 root     root       13,  32 Jan  1 00:00 mouse0
root@rpidash:/etc/X11# evtest /dev/input/mice
evtest: can't get version: Inappropriate ioctl for device
root@rpidash:/etc/X11# evtest /dev/input/mouse0
evtest: can't get version: Inappropriate ioctl for device

```

merge, 2018-08-08:
well. it helps. you don't use an evdev driver. you presumably have `module_raw input` in your ts.conf file. That's wrong because that expects evdev. Please try `module_raw waveshare` instead.

MWP, 2018-08-08:
Okay, so module_raw waveshare fixed the errors, but the touchscreen doesnt appear to do anything at all.
xev doesnt show any touch inputs.

There is a binary called "tspress" which appears to be a Qt touch input test app.
It also doesnt register any touchpresses anywhere.

And....
```
root@rpidash:/etc/X11/xinit# ts_print
ts_setup: No such file or directory
root@rpidash:/etc/X11/xinit# ts_calibrate
ts_setup: No such file or directory
root@rpidash:/etc/X11/xinit# ts_verify
tslib 1.16 / libts ABI version 0 (0x000901)
Release-Date: 2018-04-19
===================== test run 0 =====================
ts_setup: No such file or directory
Segmentation fault
root@rpidash:/etc/X11/xinit# ts_test
ts_open: No such file or directory

```

How frustrating.

Thank you very much for your continued help though!!!

merge, 2018-08-09:
You'd have to find out your waveshare vid and pid. The waveshare module_raw seems to __require__ to add `vid_pid=XXXX:YYYY` where XXXX is the vendor ID and YYYY is the product ID.

So try `module_raw waveshare vid_pid=0eef:0005`

merge, 2018-08-09:
oh. and also the "data length" seems to be required: the ts.conf syntax is:

`module_raw waveshare vid_pid=<vendorID>:<productID> len=<raw_data_len>` and I'd try
`module_raw waveshare vid_pid=0eef:0005 len=25`

merge, 2018-09-04:
please feel free to open a new issue about using `module_raw waveshare` in tslib, in case that still doesn't work for you. I just clean up. thanks!

closed 2018-09-04 by merge

tparvais, 2020-01-19:
Hello
I try to run Waveshare 7" hdmi (C) on picoreplayer (tinycore) with TSLIB 1.16 compiled for this system.

`tc@piCorePlayer:~$ cat /proc/bus/input/devices
I: Bus=0003 Vendor=0eef Product=0005 Version=0110
N: Name="BYZHYYZHY By ZH851 Mouse"
P: Phys=usb-20980000.usb-1.3.3/input0
S: Sysfs=/devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.0/0003:0EEF:0005.0001/input/input0
U: Uniq=c0U585P865
H: Handlers=mouse0 event0 js0
B: PROP=0
B: EV=1b
B: KEY=70000 0 0 0 0 0 0 0 0
B: ABS=3
B: MSC=10

I: Bus=0003 Vendor=0eef Product=0005 Version=0110
N: Name="BYZHYYZHY By ZH851 Touchscreen"
P: Phys=usb-20980000.usb-1.3.3/input0
S: Sysfs=/devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.0/0003:0EEF:0005.0001/input/input1
U: Uniq=c0U585P865
H: Handlers=mouse1 event1
B: PROP=0
B: EV=1b
B: KEY=401 0 1 0 0 0 0 0 0 0 0
B: ABS=100 3
B: MSC=10
`

When I use ts_calibrate with event1, the calibration does not work: crosshair is well displayed, bt when I press top-left corner, it jumps immediately to bottom right or further. at the end the calibration file is not correct ad the touchpad does not work.

`# Access plugins
################

# Uncomment if you wish to use the linux input layer event interface
#module_raw input

module_raw waveshare vid_pid=0eef:0005 len=25

# For other driver modules, see the ts.conf man page

# Filter plugins
################

# Uncomment if first or last samples are unreliable
# module skip nhead=1 ntail=1

# Uncomment if needed for devices that measure pressure
module pthres pmin=1

# Uncomment if needed
module debounce drop_threshold=40

# Uncomment if needed to filter spikes
# module median depth=5

# Uncomment to enable smoothing of fraction N/D
# module iir N=6 D=10

# Uncomment if needed
# module lowpass factor=0.1 threshold=1

# Uncomment if needed to filter noise samples
module dejitter delta=100

# Uncomment and adjust if you need to invert an axis or both
# module invert x0=800 y0=480

# Uncomment to use ts_calibrate's settings
module linear
`
I tried the debounce mode, no success.

If i tried module_raw waveshare vid_pid=0eef:0005 len=25 i obtained ts_read bad file descriptor.

I'm stuck completly. Don't know if this is a bug or a wrong config.

Thank you