libts/tslib #140: No release event

disaderp, 2018-10-11:
architecture and operating system: debian jessie armhf
the version of tslib: 1.17-1
the ts.conf configuration: default configuration

I have a goodix gt80x touchscreen. I wanted to setup xorg to use it in my mini pc. 
I started with default evdev driver, but I got this error: ````attempted to start touch without x/y````. 
I installed tslib and ran ````ts_calibrate````. In this utility the touchscreen was properly detected and I could finish the calibration.

I decided to use tslib xorg input driver, I built 1.1.0 version of xf86-input-tslib and installed it.
The touchscreen was detected and no errors were thrown, but there is no release event. After first touch, each next touch is just a drag starting from the initial touch spot. There is nothing I can do to release the first touch. 
I also tried using ````ts_uinput```` daemon, but the result is the same. 

I wanted to check if the problem with releasing exists only in xorg environment, so I ran ````ts_test```` with unfiltered and filtered device. The test utility was working okay with unfiltered device, I could easily switch modes and draw something on screen consisting of many separate lines. However when I ran test utility with filtered device from ````ts_uinput```` I couldn't even switch modes, I could only drag the crosshair, even when I released my finger, each next touch was also a drag, that's why I coudn't switch mode (to draw for example).

Here is output of ````evtest```` (I touched the screen a few times):
unfiltered:
````
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0xdead product 0xbeef version 0x1105
Input device name: "gt80x"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 16 (ABS_HAT0X)
      Value      0
      Min        0
      Max        0
    Event code 18 (ABS_HAT1X)
      Value      0
      Min        0
      Max        0
    Event code 21 (ABS_HAT2Y)
      Value      0
      Min        0
      Max        0
    Event code 22 (ABS_HAT3X)
      Value      0
      Min        0
      Max        0
    Event code 25 (ABS_DISTANCE)
      Value      0
      Min        0
      Max        0
    Event code 48 (ABS_MT_TOUCH_MAJOR)
      Value      0
      Min        0
      Max      255
    Event code 50 (ABS_MT_WIDTH_MAJOR)
      Value      0
      Min        0
      Max      255
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max      800
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max      480
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max        5
Properties:
Testing ... (interrupt to exit)
Event: time 1539184616.306383, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 399
Event: time 1539184616.306402, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 155
Event: time 1539184616.306415, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
[... 256 lines cut ...]
Event: time 1539184835.789553, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 27
Event: time 1539184835.789557, type 3 (EV_ABS), code 50 (ABS_MT_WIDTH_MAJOR), value 27
Event: time 1539184835.789569, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 1539184835.789572, -------------- EV_SYN ------------
Event: time 1539184835.804767, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 585
Event: time 1539184835.804772, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 249
Event: time 1539184835.804778, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 26
Event: time 1539184835.804781, type 3 (EV_ABS), code 50 (ABS_MT_WIDTH_MAJOR), value 26
Event: time 1539184835.804794, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 1539184835.804797, -------------- EV_SYN ------------
Event: time 1539184835.820000, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 586
Event: time 1539184835.820005, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 249
Event: time 1539184835.820011, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 19
Event: time 1539184835.820014, type 3 (EV_ABS), code 50 (ABS_MT_WIDTH_MAJOR), value 19
Event: time 1539184835.820026, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 1539184835.820030, -------------- EV_SYN ------------
Event: time 1539184835.835014, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 586
Event: time 1539184835.835019, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 249
Event: time 1539184835.835024, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 0
Event: time 1539184835.835028, type 3 (EV_ABS), code 50 (ABS_MT_WIDTH_MAJOR), value 0
Event: time 1539184835.835040, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 1539184835.835043, -------------- EV_SYN ------------
````

I would really appreciate any help.

merge, 2018-10-11:
thanks for this report. Your only pen-up/down symbol is ABS_MT_TOUCH_MAJOR and ABS_MT_WIDTH_MAJOR. The input module_raw doesn't support that yet.

I'll keep this on my radar for the next few days. You could help if you want to: Do a recording of a few touches with `evemu-record` of the original device. (or write it yourself and put up a pull request for review :)

p.s.: using the ts_uinput device as the input device for ts_test_mt is not really supported. This applies tslib's filter two times.

disaderp, 2018-10-11:
Thank you for your reply

I recorded a few touches and a longer slide: [https://gist.github.com/disaderp/b7e3502c711fd9aa812c8a2d8fe8c0c3](url)

merge, 2018-10-15:
If you can, please test the current master branch (or wait for an 1.18-rc2 release that makes it easier for you).

This device is a bit weird -.- it indicates to have 6-point multitouch by it's tracking-id values but that's not really standard-conform and not supported by tslib right now. Does any other Linux driver support multitouch for this? So while we should now have correct pen-up/down, we can't easily support multitouch for this device...

closed 2018-11-21 by merge