libts/tslib #141: ts_uinput

MVBeag, 2018-12-04:
Hello,

I am using Yocto through (https://github.com/digi-embedded) to develop a interface between the imx6ulsbc and Touch (FT5426). 

Is a multi touch screen so at the beginning I want to calibrate it, but the open embedded version added tslib_1.1.bb (https://layers.openembedded.org/layerindex/recipe/21187/) whitout MT --> ERROR.
After I decide to add the tslib_1.16.bb recipe in my project (/usr/local/dey-2.4/sources/meta-openembedded/meta-oe/recipes-graphics/tslib) (https://layers.openembedded.org/layerindex/recipe/53080/) it works fine after ts_calibrate I could see the result with ts_test_mt. --> GOOD

The next step was to catch the tslib event in my Qt application. I was not able to add the right tslib (and they are single touch) plugings in Qt so I decide to use ts_uinput because of your post:
"as of tslib version 1.3, you could use the included ts_uinput daemon to create an evdev input device. Point your evdev environment to it and you'd have tslib filtered and calibrated touch input without using QT's tslib plugin."
but no ts_uinput in my project, my project -- > root@ccimx6ulsbc:/usr/bin# ls: ts_calibrate .... ts_test_mt .... but no ts_uinput. Explanation:
Yocto built rootfs.ubifs ... with "bit-bake -C compile virtual/kernel" and then "bitbake -C compile dey-image-qt". Bitbake looks in different path to build the ubifs files and in /ccimx6ulsbc/tmp/work/cortexa7hf-neon-dey-linux-gnueabi/tslib/1.16-r0/tslib-1.16/tools/ts_uinput.c the file existe so what goes wrong? Is my Linux version too old? 
For information:
root@ccimx6ulsbc:/usr/bin# uname -m
armv7l
root@ccimx6ulsbc:/usr/bin# uname -r
4.9.81-dey+g4b330d1

MVBeag, 2018-12-05:
Hello Martin,

thanks for the answer, I found in dey-image-qt-fb-ccimx6ulsbc-20181122101329.rootfs.manifest (the build information):
tslib-calibrate cortexa7hf_neon 1.16
tslib-conf cortexa7hf_neon 1.16
tslib-tests cortexa7hf_neon 1.16
in parallel I did a voegelm@voegelm-desktop:~/workspace/ccimx6ulsbc$ grep -r -i "ts_uinput"
to try to find if I have a error message in the integration of the tslib recipes (similar to makefiles in GNU) but nothing. --> I will continue to prospect that
No module variance.
I wanted to switch to 1.18 but open embedded didn't develop a recipe for that: https://layers.openembedded.org/layerindex/recipe/53080/

pfink-christ, 2018-12-28:
I can confirm that I have a working yocto setup with tslib 1.15 and ts_uinput.
> I wanted to switch to 1.18 but open embedded didn't develop a recipe for that:
As Martin said, you can easily do this yourself.
Did you set a RDEPENDS on "tslib-uinput" somewhere in your image or include it in IMAGE_INSTALL? That could be the reason that it is not built in the first place.
Are you still facing this issue or did you already solve it?

merge, 2019-01-07:
can you include a working example recipe in here?

pfink-christ, 2019-09-20:
sorry - a bit late to the party, but here we go:

```
SUMMARY = "An abstraction layer for touchscreen panel events"
DESCRIPTION = "Tslib is an abstraction layer for touchscreen panel \
events, as well as a filter stack for the manipulation of those events. \
Tslib is generally used on embedded devices to provide a common user \
space interface to touchscreen functionality."
HOMEPAGE = "http://tslib.org/"

AUTHOR = "Martin Kepplinger <martink@posteo.de>"
SECTION = "base"
LICENSE = "LGPLv2+ & GPLv2+"
LIC_FILES_CHKSUM = "\
    file://COPYING;md5=fc178bcd425090939a8b634d1d6a9594 \
    file://tests/COPYING;md5=a23a74b3f4caf9616230789d94217acb \
"

SRC_URI = "https://github.com/kergoth/tslib/releases/download/${PV}/tslib-${PV}.tar.xz;downloadfilename=tslib-${PV}.tar.xz \
           file://ts.conf \
           file://tslib.sh \
"
SRC_URI[md5sum] = "9ff36788c9ab76a10b58422227b9629e"
SRC_URI[sha256sum] = "7ce48807cab655076d71a1ceef3ed0ab8a25df98074981d4a8fd1477ee5f710c"

UPSTREAM_CHECK_URI = "https://github.com/kergoth/tslib/releases"

inherit autotools pkgconfig

PACKAGECONFIG ??= "debounce dejitter iir linear median pthres skip lowpass invert variance input touchkit waveshare"
PACKAGECONFIG[debounce] = "--enable-debounce,--disable-debounce"
PACKAGECONFIG[dejitter] = "--enable-dejitter,--disable-dejitter"
PACKAGECONFIG[iir] = "--enable-iir,--disable-iir"
PACKAGECONFIG[linear] = "--enable-linear,--disable-linear"
PACKAGECONFIG[median] = "--enable-median,--disable-median"
PACKAGECONFIG[pthres] = "--enable-pthres,--disable-pthres"
PACKAGECONFIG[skip] = "--enable-skip,--disable-skip"
PACKAGECONFIG[lowpass] = "--enable-lowpass,--disable-lowpass"
PACKAGECONFIG[invert] = "--enable-invert,--disable-invert"
PACKAGECONFIG[variance] = "--enable-variance,--disable-variance"
PACKAGECONFIG[input] = "--enable-input,--disable-input"
PACKAGECONFIG[tatung] = "--enable-tatung,--disable-tatung"
PACKAGECONFIG[touchkit] = "--enable-touchkit,--disable-touchkit"
PACKAGECONFIG[waveshare] = "--enable-waveshare,--disable-waveshare"
PACKAGECONFIG[ucb1x00] = "--enable-ucb1x00,--disable-ucb1x00"
PACKAGECONFIG[mk712] = "--enable-mk712,--disable-mk712"
PACKAGECONFIG[h3600] = "--enable-h3600,--disable-h3600"
PACKAGECONFIG[dmc] = "--enable-dmc,--disable-dmc"
PACKAGECONFIG[linear-h2200] = "--enable-linear-h2200,--disable-linear-h2200"
PACKAGECONFIG[corgi] = "--enable-corgi,--disable-corgi"
PACKAGECONFIG[collie] = "--enable-collie,--disable-collie"
PACKAGECONFIG[arctic2] = "--enable-arctic2,--disable-arctic2"
PACKAGECONFIG[dmc_dus3000] = "--enable-dmc_dus3000,--disable-dmc_dus3000"
PACKAGECONFIG[cy8mrln-palmpre] = "--enable-cy8mrln-palmpre,--disable-cy8mrln-palmpre"
PACKAGECONFIG[galax] = "--enable-galax,--disable-galax"
PACKAGECONFIG[debug] = "--enable-debug,--disable-debug"

do_install_prepend() {
    install -m 0644 ${WORKDIR}/ts.conf ${S}/etc/ts.conf
}

do_install_append() {
    install -d ${D}${sysconfdir}/profile.d/
    install -m 0755 ${WORKDIR}/tslib.sh ${D}${sysconfdir}/profile.d/
}

RPROVIDES_tslib-conf = "libts-0.0-conf"

PACKAGES =+ "tslib-conf tslib-tests tslib-calibrate tslib-uinput"
DEBIAN_NOAUTONAME_tslib-conf = "1"
DEBIAN_NOAUTONAME_tslib-tests = "1"
DEBIAN_NOAUTONAME_tslib-calibrate = "1"
DEBIAN_NOAUTONAME_tslib-uinput = "1"

RDEPENDS_${PN} = "tslib-conf"
RRECOMMENDS_${PN} = "pointercal"

FILES_${PN}-dev += "${libdir}/ts/*.la"
FILES_tslib-conf = "${sysconfdir}/ts.conf ${sysconfdir}/profile.d/tslib.sh ${datadir}/tslib"
FILES_${PN} = "${libdir}/*.so.* ${libdir}/ts/*.so*"
FILES_tslib-calibrate += "${bindir}/ts_calibrate"
FILES_tslib-uinput += "${bindir}/ts_uinput"

FILES_tslib-tests = "${bindir}/ts_harvest ${bindir}/ts_print ${bindir}/ts_print_raw ${bindir}/ts_print_mt \
                     ${bindir}/ts_test ${bindir}/ts_test_mt ${bindir}/ts_verify ${bindir}/ts_finddev"
```

emmaperea, 2020-03-03:
> Hello,
>
> I am using Yocto through (https://github.com/digi-embedded) to develop a interface between the imx6ulsbc and Touch (FT5426).
>
> Is a multi touch screen so at the beginning I want to calibrate [...]

> Hello,
>
> I am using Yocto through (https://github.com/digi-embedded) to develop a interface between the imx6ulsbc and Touch (FT5426).
>
> Is a multi touch screen so at the beginning I want to calibrate [...]
Hello, 
Sorry to bother you, I don't know if It is okay to ask here but I wanted to ask if you used tslib. I am working with an imx6sbc digi distribution, using dey 2.4 as you. 
I have a problem with my app, in a certain time when I press touchscreen the app freezes, I am almost sure that the problem is Xlib that lock display for some reason, so one of the alternatives that I am considering is to use tslib in my project.   
Thanks for your time.

emmaperea, 2020-03-03:
Hello,
Sorry to bother you, I don't know if It is okay to ask here but I wanted to ask if you used tslib. I am working with an imx6sbc digi distribution, using dey 2.4 as you.
I have a problem with my app, in a certain time when I press touchscreen the app freezes, I am almost sure that the problem is Xlib that lock display for some reason, so one of the alternatives that I am considering is to use tslib in my project.
Thanks for your time.

MVBeag, 2020-03-04:
Hello,
it was for a long time ago, I forgot the idea to use this library.I can ask someone around me.
BR

-----E-mail d'origine-----

Envoyé le: Ma, 3 Mar 2020 15:13

emmaperea, 2020-03-09:
Ok thanks, if you can ask someone how could use tslib.
Thanks
Emma