libts/tslib #144: ts_calibrate: "ts_setup: No such file or directory"

Spanni26, 2019-01-08:
Hello,

I try to calibrate a embedded system with opensuse leap. I installed tslib 1.16 from the opensuse buidl service.

When I try to calibrate the touch with "ts_calibrate" I only get "ts_setup: No such file or directory"

ts.conf is barely standard:
module_raw input
module pthres pmin=1
module dejitter delta=100
module linear

xinput shows as input device:
Reakin TS2005F-USB

Any ideas?

Regards

Daniel

merge, 2019-01-08:
Seems like ts_setup() doesn't find one, and falls back to /dev/input/event0 which doesn't work... what list of input devices does `evemu-record` or `evtest` give you?

Spanni26, 2019-01-08:
Hello,

evtest shows: 
 aaz0aidu:~ # evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Reakin TS2005F-USB
/dev/input/event1:      Microsoft Comfort Curve Keyboard 3000
/dev/input/event2:      Microsoft Comfort Curve Keyboard 3000
/dev/input/event3:      Power Button
/dev/input/event4:      Sleep Button
/dev/input/event5:      Power Button
/dev/input/event6:      PC Speaker
/dev/input/event7:      HDA Intel HDMI/DP,pcm=3
/dev/input/event8:      HDA Intel Mic
/dev/input/event9:      HDA Intel Line
/dev/input/event10:     HDA Intel Headphone

Regards

Daniel

merge, 2019-01-08:
have you verified your setup according to our docs? Try to `export TSLIB_TSDEVICE=/dev/input/event0` before starting ts_calibrate. If that works you can find a more elegant way to tell tslib what your touchscreen is.

Spanni26, 2019-01-08:
Ok, with the exported TSLIB_DEVICE the calibration works....
Thanks.

Now, I have to find out how to get it to work at boot time....

Regards

Daniel

merge, 2019-01-08:
tslib doesn't automatically find your touchscreen device. the kernel driver might not be complete enough or buggy. so _you_ need to "find" it at boot time. however you want (driver name for example).

then you can create a symlink called `/dev/input/ts` that points to the found event device; or, just like now, set TSLIB_TSDEVICE to the found device; the result is the same.

Spanni26, 2019-01-08:
Ok, maybe I'm missing something...
My Udev now creates /dev/input/ts, which is a symlink to the event0.
ts_calibration now works out_of_the_box. But how do I configure the system to use the calibration?

merge, 2019-01-09:
well. read the README. I recommend using ts_uinput, i.e. starting `ts_uinput -d -v` at startup and read the input device you need to point your system (X11) to, to use. We include an [example script](https://github.com/kergoth/tslib/blob/master/tools/ts_uinput_start.sh).

Spanni26, 2019-01-09:
Started ts_uinput via systemd. 
A new /dev/input/event?  exists, it is recognized by Xorg (according to log) but the calibration seems to have no effect....

merge, 2019-01-14:
actually I'm not sure how to do it in X11, but I'm sure you'll find out how to force your evdev xf86 input driver to use the new event device...

merge, 2019-01-23:
as the original issue is actually handled, I guess we'll close this. Please feel free to open an issue about how to use ts_uinput in X11, or any other problem you might have!

thanks!

closed 2019-01-23 by merge

urus-vp, 2020-01-26:
Hello!
I have the same issue... I just tried to implement your recomendation above, but w/o success...
please, take a look at the attached screenshots.
Do you have any ideas ?

note:
there are three input devices... but I can get events from event2 only

[image]
[image]