libts/tslib #148: invalid coordinates on release

DerDakon, 2019-03-27:
The linear plugin will set the coordinates of a release event to 0, 0. Afterwards the calibration will be applied to it, so these coordinates become "random" (from a users perspective). For me it usually was something like -11, -9. This breaks e.g. the missing coordinates detection in [Qt](https://github.com/qt/qtbase/blob/63156e20629a087a761cbe34b110681267d58a93/src/platformsupport/input/tslib/qtslib.cpp#L118) in all versions up to 5.12.2 [fix in review](https://codereview.qt-project.org/#/c/257062/). Maybe the calibration should be guarded by something like this:

```c
if (pressure != 0 || samp->x != || samp->y !=0)
```

merge, 2019-03-27:
what tslib modules do you use, i.e. how does your ts.conf file look like?

thanks for reporting this. maybe we shouldn't have zeros in a release event...

DerDakon, 2019-03-27:
module_raw input
module linear

Looks like it is the other plugin that set the coordinates to 0 and then linear breaks it, but anyway.

merge, 2019-03-28:
while I think about it I guess the fix in Qt in ok. This should only apply to the the legacy interface that Qt is using. If pressure is 0, you shouldn't evaluate the x/y coordinates there.

It's about time to implement `ts_read_mt()` in qtslib... (`#ifdef TSLIB_VERSION_MT`).

merge, 2019-04-24:
again, thanks for reporting this. seems to be fixed in Qt.

closed 2019-04-24 by merge