libts/tslib #151: Possible memory leak in scandevices from ts_setup.c

crujose, 2019-04-25:
https://github.com/kergoth/tslib/blob/2abfc27c31ed254650f21e013c1add253a513c61/src/ts_setup.c#L92

If the file fails to open or the ioctl call fails (i.e. anywhere `continue` is used in the loop prior to `free(namelist[i]);`) then this namelist entry is never freed.

The early `return` on line 99 would leak any remaining entries in the list including the list itself (granted, this would only happen in the ironic scenario where malloc fails).

The early return at line 108 would leak all entries in namelist from `i+1` to `ndev-1`

There are a lot of ways to address this, but here's one suggestion:
```C
for (i = 0; i < ndev; ++i) {
    // Current scanning logic, but using break instead of return,
    // and without handling cleanup of namelist.
    // ...
}
for (i = 0; i < ndev; ++i) {
    free(namelist[i]);
}
free(namelist);

return filename; // Will be NULL if a device is not found (or malloc failed),
                 // otherwise it will point to a heap-allocated string containing the filename.
```

merge, 2019-04-26:
thanks. indeed your suggestion looks cleaner. could you create a pull request for this change?

crujose, 2019-04-27:
I could, but I haven't actually cloned this repo or used this library before so I'd be starting from scratch (i.e I'm lazy).   I only happened to notice `scandir` because I recently used it in a project and all of its pitfalls were still fresh in my mind.

commit 2be6a0e5 (2019-04-29): ts_setup: fix possible memory leak

Fixes #151
closed 2019-04-29 by merge