libts/tslib #152: Goodix I2c gt9XX No release event

soroush, 2019-06-11:
Hi 

I am trying to get tslib to work on a 7 inch resistive touch panel (goodix on i2c). Problem is that tslib does not generate release event (Pressure = 0 ?). Consider ts_test output: I expect when I pick up my finger, there must be an event with pressure reported as 0. like this one (which was working fine):

```
....                                                                                                       
1560198942.105126:    698     24    255                                                                                                        
1560198942.115198:    698     25    255                                                                                                        
1560198942.116254:    699     27      0   

```
Unfortunately I have no access to source code or possible patches on tslib of above OS. Mine looks like this:

```
....                                                                                                        
1560200222.632372:    283    289    255                                                                                                        
1560200222.651983:    283    289    255  
```
Application behavior in Qt side also suggests this: No button releases after touch. Just like as if you clicked the left button of mouse and never released it. I guess this might be related to [issue#140](https://github.com/libts/tslib/issues/140)

merge, 2019-06-11:
thanks for reporting. how does your ts.conf file with the modules look like?

soroush, 2019-06-11:
$ cat /etc/ts.conf

```
module_raw input

# Uncomment if you're using a Sharp Zaurus SL-5500/SL-5000d
# module_raw collie

# Uncomment if you're using a Sharp Zaurus SL-C700/C750/C760/C860
# module_raw corgi

# Uncomment if you're using a device with a UCB1200/1300/1400 TS interface
# module_raw ucb1x00

# Uncomment if you're using an HP iPaq h3600 or similar
# module_raw h3600

# Uncomment if you're using a Hitachi Webpad
# module_raw mk712

# Uncomment if you're using an IBM Arctic II
# module_raw arctic2

module pthres pmin=1
module variance delta=30
module dejitter delta=100
module linear
```

merge, 2019-06-11:
please test again using the bare minimum you need. this might be "module_raw input" only. only if you have a calibfile (ran ts_calibrate) you might "module linear" too. (at least to test).

from what I can tell: the "variance" module is deprecated and in not a very good shape. If you need a filter like that, take a look at the "median" filter.

soroush, 2019-06-11:
I disabled everything except module_raw input, there is no difference. Running `ts_test` fourth column is always 255 :

```
1560265776.873430:    586   1208    255                                                                                                        
1560265776.893292:   1018   1219    255                                                                                                        
1560265776.894422:   1035   1206    255                                                                                                        
1560265777.093441:    438   1214    255                                                                                                        
1560265777.113282:   1020   1206    255                                                                                                        
1560265777.133286:   1029   1207    255                                                                                                        
1560265777.134404:    661   2045    255                                                                                                        
1560265777.293444:    880   1125    255                                                                                                        
1560265777.313287:    893   1127    255                                                                                                        
1560265777.333615:    889   1125    255                                                                                                        
1560265777.334720:    886   1125    255  
```

soroush, 2019-06-11:
By the way this is the device tree node in case it helps:

```
    goodix_ts@5d {
        compatible = "goodix,gt9xx";
        reg = <0x5d>;
        status = "okay";
        pinctrl-0 = <&pinctrl_gt9xx>;
        interrupt-parent = <&gpio5>;
        interrupts = <1 2>;
        goodix,rst-gpio = <&gpio5 0 2>;
        goodix,irq-gpio = <&gpio5 1 2>;
        goodix,cfg-group0 = [
            42 D0 02 00 05 05 75 01 01 0F 24
            0F 64 3C 03 05 00 00 00 02 00 00
            00 16 19 1C 14 8C 0E 0E 24 00 31
            0D 00 00 00 83 33 1D 00 41 00 00
            00 00 00 08 0A 00 2B 1C 3C 94 D5
            03 08 00 00 04 93 1E 00 82 23 00
            74 29 00 69 2F 00 5F 37 00 5F 20
            40 60 00 F0 40 30 55 50 27 00 00
            00 00 00 00 00 00 00 00 00 00 00
            00 00 00 00 00 00 00 14 19 00 00
            50 50 02 04 06 08 0A 0C 0E 10 12
            14 16 18 1A 1C 00 00 00 00 00 00
            00 00 00 00 00 00 00 00 00 00 1D
            1E 1F 20 21 22 24 26 28 29 2A 1C
            18 16 14 13 12 10 0F 0C 0A 08 06
            04 02 00 00 00 00 00 00 00 00 00
            00 00 00 00 00 00 00 00 9C 01];
        goodix,cfg-group2 = [
            48 D0 02 00 05 05 75 01 01 0F 24
            0F 64 3C 03 05 00 00 00 02 00 00
            00 16 19 1C 14 8C 0E 0E 24 00 31
            0D 00 00 00 83 33 1D 00 41 00 00
            3C 0A 14 08 0A 00 2B 1C 3C 94 D5
            03 08 00 00 04 93 1E 00 82 23 00
            74 29 00 69 2F 00 5F 37 00 5F 20
            40 60 00 F0 40 30 55 50 27 00 00
            00 00 00 00 00 00 00 00 00 00 00
            00 00 00 00 00 00 00 14 19 00 00
            50 50 02 04 06 08 0A 0C 0E 10 12
            14 16 18 1A 1C 00 00 00 00 00 00
            00 00 00 00 00 00 00 00 00 00 1D
            1E 1F 20 21 22 24 26 28 29 2A 1C
            18 16 14 13 12 10 0F 0C 0A 08 06
            04 02 00 00 00 00 00 00 00 00 00
            00 00 00 00 00 00 00 00 3C 01];
    };
```

And mt tslib version is 1.19

merge, 2019-06-12:
nope. tslib reads input event from the linux driver. There are various ways to detect "release" in evdev input events (linux driver output). So What would help me is this:

Run "evemu-record /dev/input/eventX" with X being your actual touchscreen device. And to a 2 or 3 "taps" on the screen, save the output and post that here. Do you think you could do that? If it's easier for you, you can also use the "evtest" program (which is similar).

thanks for your help.

soroush, 2019-06-12:
I ran evemu-record /dev/input/event1 (/dev/input/event1 is my actual touch device) and touched the screen several times.

The result is as follows :

```
# EVEMU 1.3
# Kernel: 4.18.11-ametis
# Input device name: "iMX6UL Touchscreen Controller"
# Input device ID: bus 0x19 vendor 0000 product 0000 version 0000
# Supported events:
#   Event type 0 (EV_SYN)
#     Event code 0 (SYN_REPORT)
#     Event code 1 (SYN_CONFIG)
#     Event code 2 (SYN_MT_REPORT)
#     Event code 3 (SYN_DROPPED)
#     Event code 4 ((null))
#     Event code 5 ((null))
#     Event code 6 ((null))
#     Event code 7 ((null))
#     Event code 8 ((null))
#     Event code 9 ((null))
#     Event code 10 ((null))
#     Event code 11 ((null))
#     Event code 12 ((null))
#     Event code 13 ((null))
#     Event code 14 ((null))
#     Event code 15 (SYN_MAX)
#   Event type 1 (EV_KEY)
#     Event code 330 (BTN_TOUCH)
#   Event type 3 (EV_ABS)
#     Event code 0 (ABS_X)
#       Value     1593
#       Min          0
#       Max       4095
#       Fuzz         0
#       Flat         0
#       Resolution   0
#     Event code 1 (ABS_Y)
#       Value     1815
#       Min          0
#       Max       4095
#       Fuzz         0
#       Flat         0
#       Resolution   0
# Properties:
N: iMX6UL Touchscreen Controller
I: 0019 0000 0000 0000
P: 00 00 00 00 00 00 00 00
B: 00 0b 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 04 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 02 00 00 00 00 00 00 00 00
B: 03 03 00 00 00 00 00 00 00
B: 04 00 00 00 00 00 00 00 00
B: 05 00 00 00 00 00 00 00 00
B: 11 00 00 00 00 00 00 00 00
B: 12 00 00 00 00 00 00 00 00
B: 14 00 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
[... 174 lines cut ...]
E: 5.209899 0003 0001 3516	# EV_ABS / ABS_Y                3516
E: 5.209899 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +20ms
E: 5.230077 0003 0000 0741	# EV_ABS / ABS_X                741
E: 5.230077 0003 0001 3519	# EV_ABS / ABS_Y                3519
E: 5.230077 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +21ms
E: 5.249908 0003 0000 0742	# EV_ABS / ABS_X                742
E: 5.249908 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +19ms
E: 5.289996 0003 0001 3518	# EV_ABS / ABS_Y                3518
E: 5.289996 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +40ms
E: 5.309911 0003 0000 0738	# EV_ABS / ABS_X                738
E: 5.309911 0003 0001 3515	# EV_ABS / ABS_Y                3515
E: 5.309911 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +20ms
E: 5.329897 0003 0000 0743	# EV_ABS / ABS_X                743
E: 5.329897 0003 0001 3519	# EV_ABS / ABS_Y                3519
E: 5.329897 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +20ms
E: 5.350026 0003 0001 3518	# EV_ABS / ABS_Y                3518
E: 5.350026 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +21ms
E: 5.369911 0003 0000 0753	# EV_ABS / ABS_X                753
E: 5.369911 0003 0001 3511	# EV_ABS / ABS_Y                3511
E: 5.369911 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +19ms
E: 5.371022 0003 0000 0758	# EV_ABS / ABS_X                758
E: 5.371022 0003 0001 3501	# EV_ABS / ABS_Y                3501
E: 5.371022 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +2ms

```

merge, 2019-06-13:
do you have the sources of your linux? do you have a second piece of hardware to test? this looks like a Linux or Hardware problem to me. BTN_TOUCH should be sent, according  to the driver but that's missing for you...

soroush, 2019-06-15:
Hi

Sorry for late reply. I was struggling with i.MX TSC kernel driver. You are right, it's a kernel problem, initiated by poor hardware design. The genius behind hardware design, decided to connect 4 analog signals over a long track with 4 vias + 4 fpc connectors + 1 b2b connector. And also, he or she decided not to connect touch detect signal for the TSC using a 5th GPIO pin, but used XNUR analog signal as touch detection. It does not work obviously (randomly works, with a very bad probability).

For those who may encounter same problem on i.MX 6UL boards, and found this by search:
following workaround solves the problem. Change 2ms pre-charge time to 5 or 10 milliseconds here: 

https://github.com/torvalds/linux/blob/0011572c883082a95e02d47f45fc4a42dc0e8634/drivers/input/touchscreen/imx6ul_tsc.c#L243

Its effect on user experience looks negligible.

There is another path by boundrydevices which does not work on newer version of kernel:

https://github.com/boundarydevices/linux-imx6/blob/1e0782c89c2438b1aabb64ab7ef3e8e4d8634fea/drivers/input/touchscreen/imx6ul_tsc.c#L237

closed 2019-06-15 by soroush