libts/tslib #153: How to check if there are pending events when tslib opened in blocking mode

guillerodriguez, 2019-07-09:
I need to open tslib in blocking mode, but need a way to check if there are outstanding events. Basically I need to:

- Block until at least one event is ready. This part is easy
- But if there are many events ready, discard all but the most recent one. This is the way I don't know how to handle.

I would try to get the fd using `ts_fd` and then use select or poll, like this: https://github.com/nicupavel/android-tslib/blob/master/tests/ts_calibrate.c#L166

However #53 suggests that this is not supported.

What is the proper way to do it then?

merge, 2019-07-09:
in blocking mode, ts_read_mt() will return if there's an event (or the number of events you want to read at once). I might get you wrong, but why would you want to lose events? This can result in bad user experience, can't it?

guillerodriguez, 2019-07-09:
I have a touch controller that reports events at a very high rate (more than 200 per second). This is creating an unnecessary load in the system so I am trying to filter the event stream and limit the max rate to e.g. 60 Hz.

My current approach for this is to have the app force a minimum delay of 16.7ms between reads. If there are no pending events, the read call should block (this works). But if there are more than pending events, I am only interested in the last one, and want to skip the others.

merge, 2019-07-09:
if poll() or select() doesn't work on the original file-descriptor, ts_fd(), then we might have a problem. But still, what if you skip "critical" samples that mess up the UX?

guillerodriguez, 2019-07-09:
select() on the fd returned by ts_fd() does certainly work. I am just slightly concerned because of your comments in #53:

> select()ing on ts_fd and then reading samples from tslib is simply wrong usage

That's why I was asking, if this is wrong, then what's the proper way to do what I need.

Re. skipping critical samples, I don't see how would that happen. The last event you look at always contains the full state -- either pressed (plus coordinates) or not (note that I'm not doing multitouch).

The only thing that could happen is if the user manages to touch, then release, all within a 16.7ms window, and then the touch goes unnoticed. But that's rather unusual I think, and something I can perfectly live with (in fact if the touch controller was reporting events at 60Hz instead of 200+ Hz, the result would be exactly the same).

merge, 2019-07-10:
> select() on the fd returned by ts_fd() does certainly work. I am just slightly concerned because of your comments in #53:
> 
> > select()ing on ts_fd and then reading samples from tslib is simply wrong usage
> 
> That's why I was asking, if this is wrong, then what's the proper way to do what I need.

IIRC that was (just) because the fd doesn't belong to libts. You actually _can_ use it, if you carefully choose the filter modules you apply in between it (the fd of the original device) and the samples you get after libts (from ts_read_mt).

I actually think you will be fine if you just don't use the variance filter. You'd need to test.