libts/tslib #155: Installation of tslib on Deepin

shacharoz, 2019-08-18:
Do you think it is possible to install and run tslib on Deepin?
Can someone please provide instructions for a clean Linux installation (without nothing)?

I just want to ensure I am doing it correctly.

merge, 2019-08-21:
you should think about, why you need tslib. Do you have a working Linux driver, i.e. you touchscreen works with Linux, but not well enough (you want to apply filters), then your "process 3" should install everything that's needed (mainly, "ts_uinput"). ts_calibrate is only necessary, if you use the "linear" tslib filter (in your ts.conf).

does your touchscreen currently not work at all with Linux?

Please post, what "evemu-record" or "evtest" tell you, when selecting the touchscreen input device.

Also, please post, what ts.conf tslib config file you currently use.

thanks

shacharoz, 2019-08-21:
thanks for the guiding questions, @merge . 

WHY TSLIB (and my touchscreen situation):
my touchscreen works with one touch point, and in Chromium i actually have multitouch gestures. since Chromium is written is c/cpp, i understood i need something in that language to understand the touch points.   -- Enter tslib (right?).

--------------------------------

for `lsusb` command i get:
> Bus 001 Device 003: ID 0eef:0005 D-WAV Scientific Co

--------------------------------

both `evtest` and `evemu-record` were not installed on my system.
after installation, the result is:
>Available devices:
/dev/input/event0:	Sleep Button
/dev/input/event1:	Power Button
/dev/input/event2:	Power Button
/dev/input/event3:	Video Bus
/dev/input/event4:	 USB OPTICAL MOUSE
/dev/input/event5:	WaveShare WS170120
/dev/input/event6:	  USB Keyboard
/dev/input/event7:	  USB Keyboard System Control
/dev/input/event8:	  USB Keyboard Consumer Control
/dev/input/event9:	HDA Intel PCH Mic
/dev/input/event10:	HDA Intel PCH Front Headphone
/dev/input/event11:	HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event12:	HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event13:	HDA Intel PCH HDMI/DP,pcm=8
/dev/input/event14:	HDA Intel PCH HDMI/DP,pcm=9
/dev/input/event15:	HDA Intel PCH HDMI/DP,pcm=10"

This is my device: "/dev/input/event5:  WaveShare WS170120"

--------------------------------

regarding ts.conf:
havent change the original installation. i have only one file in the /etc/ folder. 
i guess this is what i am using (but i didnt do anything special in order to USE it). if this is something i need to do, then maybe that is what i miss.

> # Access plugins
################
# Uncomment if you wish to use the linux input layer event interface
module_raw input
# For other driver modules, see the ts.conf man page
# Filter plugins
################
# Uncomment if first or last samples are unreliable
# module skip nhead=1 ntail=1
# Uncomment if needed for devices that measure pressure
module pthres pmin=1
# Uncomment if needed
# module debounce drop_threshold=40
# Uncomment if needed to filter spikes
# module median depth=5
# Uncomment to enable smoothing of fraction N/D
# module iir N=6 D=10
# Uncomment if needed
# module lowpass factor=0.1 threshold=1
# Uncomment if needed to filter noise samples
module dejitter delta=100
# Uncomment and adjust if you need to invert an axis or both
# module invert x0=800 y0=480
# Uncomment to use ts_calibrate's settings
module linear

-------------------------------------

do you have any recommended Ubuntu version for me?
do they differ in multitouch aspects or Java development aspects?
Kubuntu vs Lubuntu vs Ubuntu Mate for example.

merge, 2019-08-21:
> thanks for the guiding questions, @merge .
>
> WHY TSLIB (and my touchscreen situation):
> my touchscreen works with one touch point, and in Chromium i actually have multitouch gestures. since Chromium [...]

tslib won't give you multitouch if Linux doesn't support it for your device. What application uses multitouch doesn't matter. Please just test it using evtest or evemu-record, see below.

> for `lsusb` command i get:
>
>
> both `evtest` and `evemu-record` were not installed on my system.
> after installation, the result is:
>
>
> This is my device: "/dev/input/event5: WaveShare WS170120"

And after choosing it, what is the output?

shacharoz, 2019-08-21:
sorry just discovered this, attached the input file.
it seems working and responding to touch only (and not the mouse movement. i pressed with 2 fingers in this session.

[evemu-record file report.log](https://github.com/libts/tslib/files/3526137/evemu-record.file.report.log)

merge, 2019-08-21:
so you didn't move the fingers there, right? Looks like Linux sees multitouch just fine. you should be able to use that in your application. IMO you won't need tslib. You only do in case you want to add some additional filters onto this current touch input behaviour, but you'll see that when using your application... just figure out how to use the Linux input system.

if you teally want to test tslib anyways, read the docs carefully. Try to set TSLIB_TSDEVICE variable to your input device file, and comment-out _all_ "module"s in ts.conf except for the first "module_raw input" and run ts_test_mt.

shacharoz, 2019-08-21:
restarted the PC. tried again. few questions.

1. how can you tell the number of fingers from the log? i attach another one with several. where do you see that? can you point me to a guide to learn about it? 

[evemu-record multifinger report.log](https://github.com/libts/tslib/files/3526213/evemu-record.multifinger.report.log)

------------------------------------------

2. every time i restart the pc, the "Bus port" in evemu-record changes. is it normal?
since i have a feeling its going to be a software challenge to track that specific data.

------------------------------------------

3. in Java software I dont even catch one Touch event. this supposed to work definitely (and also works on Windows). that's why i was sure i have another issue.

merge, 2019-08-22:
> restarted the PC. tried again. few questions.
>
>     1. how can you tell the number of fingers from the log? i attach another one with several. where do you see that? can you point me to a guide to [...]

4 fingers. you count the slots that get assigned a tracking id, see https://www.kernel.org/doc/Documentation/input/multi-touch-protocol.txt

> 
>     1. every time i restart the pc, the "Bus port" in evemu-record changes. is it normal?
>        since i have a feeling its going to be a software challenge to track that specific data.
> 

not sure what you mean by bus port, but Linux enumerates devices randomly. You always have to parse a whole directory ( in /dev/input for example) to find your device, if you really need to use that directly.

> 
>     1. in Java software I dont even catch one Touch event. this supposed to work definitely (and also works on Windows). that's why i was sure i have another issue.

I don't how what you're writing and what APIs you use, so I can't say anything about that.

shacharoz, 2019-08-22:
Got it. So as long as I get multitouch points from the evemu-record, there is nothing else i can get from tslib?

I thought that it will give me the touch points in C as an array every frame, and then I can wrap it with JNI so that I can access this in Java. 

Currently I'm writing with javafx. 
All the examples in this page, as well as others, do not work. the TouchEvent handler is not called at all. When I touch the screen, a MouseEvent hanlder is called instead (for one finger).
https://docs.oracle.com/javafx/2/events/gestures.htm

merge, 2019-08-22:
> Got it. So as long as I get multitouch points from the evemu-record, there is nothing else i can get from tslib?

you just get it in a different format (plus the filters applied, but I think you don't need any)

> 
> I thought that it will give me the touch points in C as an array every frame, and then I can wrap it with JNI so that I can access this in Java.

sure you can try that. point tslib to your touchscreen (TSLIB_TSDEVICE) and use it's interface according to the docs.

> 
> Currently I'm writing with javafx.
> All the examples in this page, as well as others, do not work. the TouchEvent handler is not called at all. When I touch the screen, a MouseEvent hanlder is called instead (for one finger).
> https://docs.oracle.com/javafx/2/events/gestures.htm

ok. We would need to look at javafx and how they are detecting devices and read events. Is source available? Might be a bug on their side...

shacharoz, 2019-08-22:
alright. thank you a lot @merge . your feedback helps.
just to be organized for the future people, let me first close this issue with an answer and reopen regarding API with Java - because i found also one more possible way there.

shacharoz, 2019-08-22:
**ANSWER**
so, if anyone installs any Linux version and would like to make sure their system identifies multi touch points on their USB connected touch screen, you do not need to install TSLIB for this.
you can write these commands in your terminal:

`sudo apt-get install evemu-tools`
`evemu-record`

then you would a list of devices that look like this:
> /dev/input/event5: WaveShare WS170120
find your device according to its name of manufacturer. this is mine as example.

then you click the number of the input from the line: /input/event**XX**
in this case it is '5'

now the PC is waiting to receive any input from that device. touch anywhere on your screen, and data should be printed out in the terminal.
this [documentation  ](https://www.kernel.org/doc/Documentation/input/multi-touch-protocol.txt) has some explanation on how to interpret the data and understand if the system sees one touch or more.

TSLIB exposes this data (+extras) through C code.

this script was tested on Ubuntu MATE 18.04 , KUbuntu 18.04 , DeepIn.

closed 2019-08-22 by shacharoz