libts/tslib #166: Calibration issue with WaveShare 7" HDMI-LCD (C)

tparvais, 2020-01-19:
Hello
I try to run Waveshare 7" hdmi (C) on picoreplayer (tinycore) with TSLIB 1.16 compiled for this system.

`tc@piCorePlayer:~$ cat /proc/bus/input/devices
I: Bus=0003 Vendor=0eef Product=0005 Version=0110
N: Name="BYZHYYZHY By ZH851 Mouse"
P: Phys=usb-20980000.usb-1.3.3/input0
S: Sysfs=/devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.0/0003:0EEF:0005.0001/input/input0
U: Uniq=c0U585P865
H: Handlers=mouse0 event0 js0
B: PROP=0
B: EV=1b
B: KEY=70000 0 0 0 0 0 0 0 0
B: ABS=3
B: MSC=10

I: Bus=0003 Vendor=0eef Product=0005 Version=0110
N: Name="BYZHYYZHY By ZH851 Touchscreen"
P: Phys=usb-20980000.usb-1.3.3/input0
S: Sysfs=/devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.0/0003:0EEF:0005.0001/input/input1
U: Uniq=c0U585P865
H: Handlers=mouse1 event1
B: PROP=0
B: EV=1b
B: KEY=401 0 1 0 0 0 0 0 0 0 0
B: ABS=100 3
B: MSC=10
`

When I use ts_calibrate with event1, the calibration does not work: crosshair is well displayed, bt when I press top-left corner, it jumps immediately to bottom right or further. at the end the calibration file is not correct ad the touchpad does not work.

`#module_raw input

module_raw waveshare vid_pid=0eef:0005 len=25

module pthres pmin=1

module debounce drop_threshold=40

module dejitter delta=100

module linear

`
I tried the debounce mode, no success.

If i tried module_raw waveshare vid_pid=0eef:0005 len=25 i obtained ts_read bad file descriptor.

I'm stuck completly. Don't know if this is a bug or a wrong config.

Thank you

_Originally posted by @tparvais in https://github.com/libts/tslib/issues/135#issuecomment-576016658_

merge, 2020-01-20:
first of all, please provide the (full) output of either evtest or evemu-record when you do one "tap" on the screen.

what I suspect is that you should use "module_raw input" instead of "module_raw waveshare..."

tparvais, 2020-01-20:
Hello
I've tried also module_raw input initially, same issue. That's why I change to waveshare.
how an I launch evtest & evemu-record ? is that part of standard tslib binaries ? (I'm using Picoreplayer on tincore)

tparvais, 2020-01-20:
Hello, by chance evtest is available as a package. Here is the result with one tap. (and module_raw waveshare)

```
tc@piCorePlayer:~$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      BYZHYYZHY By ZH851 Mouse
/dev/input/event1:      BYZHYYZHY By ZH851 Touchscreen
Select the device event number [0-1]: 1
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0xeef product 0x5 version 0x110
Input device name: "BYZHYYZHY By ZH851 Touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
    Event code 320 (BTN_TOOL_PEN)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max     1024
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max      600
    Event code 40 (ABS_MISC)
      Value      0
      Min        0
      Max       10
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1579543253.532095, type 3 (EV_ABS), code 40 (ABS_MISC), value 3
Event: time 1579543253.532095, type 3 (EV_ABS), code 40 (ABS_MISC), value 4
Event: time 1579543253.532095, type 3 (EV_ABS), code 1 (ABS_Y), value 600
Event: time 1579543253.532095, -------------- SYN_REPORT ------------
Event: time 1579543253.548085, type 3 (EV_ABS), code 40 (ABS_MISC), value 5
Event: time 1579543253.548085, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1579543253.548085, type 3 (EV_ABS), code 1 (ABS_Y), value 0
Event: time 1579543253.548085, -------------- SYN_REPORT ------------
Event: time 1579543253.564093, type 3 (EV_ABS), code 40 (ABS_MISC), value 1
Event: time 1579543253.564093, type 3 (EV_ABS), code 0 (ABS_X), value 311
Event: time 1579543253.564093, type 3 (EV_ABS), code 1 (ABS_Y), value 237
Event: time 1579543253.564093, type 3 (EV_ABS), code 40 (ABS_MISC), value 2
Event: time 1579543253.564093, type 3 (EV_ABS), code 1 (ABS_Y), value 600
Event: time 1579543253.564093, type 3 (EV_ABS), code 40 (ABS_MISC), value 5
Event: time 1579543253.564093, -------------- SYN_REPORT ------------
Event: time 1579543253.580080, type 3 (EV_ABS), code 40 (ABS_MISC), value 3
Event: time 1579543253.580080, type 3 (EV_ABS), code 0 (ABS_X), value 0
Event: time 1579543253.580080, type 3 (EV_ABS), code 40 (ABS_MISC), value 4
Event: time 1579543253.580080, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1579543253.580080, -------------- SYN_REPORT ------------
Event: time 1579543253.596076, type 3 (EV_ABS), code 40 (ABS_MISC), value 5
Event: time 1579543253.596076, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1579543253.596076, type 3 (EV_ABS), code 1 (ABS_Y), value 0
Event: time 1579543253.596076, -------------- SYN_REPORT ------------
Event: time 1579543253.628068, type 3 (EV_ABS), code 40 (ABS_MISC), value 3
Event: time 1579543253.628068, type 3 (EV_ABS), code 40 (ABS_MISC), value 4
Event: time 1579543253.628068, type 3 (EV_ABS), code 1 (ABS_Y), value 600
Event: time 1579543253.628068, -------------- SYN_REPORT ------------
Event: time 1579543253.644076, type 3 (EV_ABS), code 40 (ABS_MISC), value 5
Event: time 1579543253.644076, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1579543253.644076, type 3 (EV_ABS), code 1 (ABS_Y), value 0
Event: time 1579543253.644076, -------------- SYN_REPORT ------------
^Ctc@piCorePlayer:~$

```

tparvais, 2020-01-20:
I succeed to get a debug-enabled tslib for picoreplayer (Thanks Ralphy). 

Here is the logs printed on the screen when launching ts_calibrate.
I observe that on my touchescreen, the Bottom-right corner step was skipped: I didn't press on it, it jumps from Top-right to Bottom left

Hope this helps.

Thank you 

```
tc@piCorePlayer:/usr/local/bin$ sudo TSLIB_FBDEVICE=/dev/fb0 TSLIB_TSDEVICE=/dev/input/event1 /usr/local/bin/ts_calibrate
tslib 1.16+ / libts ABI version 0 (0x000000)
Release-Date: [unreleased]
Host OS: Linux, trying to open /dev/input/event1
Loading module input
Loading module pthres (pmin=1)
Loading module variance (delta=30)
Loading module dejitter (delta=100)
Loading module linear
Linear calibration constants: 0 0 33554432 0 0 0 65536
xres = 1024, yres = 600
tslib input device: vendor 0xEEF product 0x5 version 0x110 on bus 0x3
RAW---------------------> 0 600 255 1579544428.624771
TS_READ_RAW----> x = 0, y = 600, pressure = 255
RAW---------------------> 0 0 255 1579544428.640763
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 30 600 255 1579544428.656760
TS_READ_RAW----> x = 30, y = 600, pressure = 255
RAW---------------------> 30 600 255 1579544428.656773
TS_READ_RAW----> x = 30, y = 600, pressure = 255
RAW---------------------> 0 0 0 1579544428.672761
TS_READ_RAW----> x = 0, y = 0, pressure = 0
Took 4 samples...
Top left : X = 15 Y = 600
RAW---------------------> 0 0 255 1579544428.688748
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 30 600 255 1579544428.704759
TS_READ_RAW----> x = 30, y = 600, pressure = 255
RAW---------------------> 0 600 255 1579544428.720755
TS_READ_RAW----> x = 0, y = 600, pressure = 255
RAW---------------------> 0 0 255 1579544428.736750
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 30 600 255 1579544428.752756
TS_READ_RAW----> x = 30, y = 600, pressure = 255
RAW---------------------> 0 600 255 1579544428.768754
TS_READ_RAW----> x = 0, y = 600, pressure = 255
RAW---------------------> 0 0 255 1579544428.784749
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 0 600 255 1579544448.704775
TS_READ_RAW----> x = 0, y = 600, pressure = 255
RAW---------------------> 0 0 255 1579544448.720761
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 1017 600 255 1579544448.736768
TS_READ_RAW----> x = 1017, y = 600, pressure = 255
RAW---------------------> 1017 600 255 1579544448.736780
TS_READ_RAW----> x = 1017, y = 600, pressure = 255
RAW---------------------> 0 0 0 1579544448.752767
TS_READ_RAW----> x = 0, y = 0, pressure = 0
Took 11 samples...
Top right : X = 0 Y = 600
RAW---------------------> 0 0 255 1579544448.768755
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 1017 600 255 1579544448.784760
TS_READ_RAW----> x = 1017, y = 600, pressure = 255
RAW---------------------> 1017 600 255 1579544448.784773
TS_READ_RAW----> x = 1017, y = 600, pressure = 255
RAW---------------------> 0 0 0 1579544448.800766
TS_READ_RAW----> x = 0, y = 0, pressure = 0
Took 3 samples...
Bot right : X = 1017 Y = 600
RAW---------------------> 0 0 255 1579544448.816757
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 1017 600 255 1579544448.832761
TS_READ_RAW----> x = 1017, y = 600, pressure = 255
RAW---------------------> 0 600 255 1579544448.848761
TS_READ_RAW----> x = 0, y = 600, pressure = 255
RAW---------------------> 0 0 255 1579544448.864754
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 1017 600 255 1579544448.880758
TS_READ_RAW----> x = 1017, y = 600, pressure = 255
RAW---------------------> 0 600 255 1579544448.896754
TS_READ_RAW----> x = 0, y = 600, pressure = 255
RAW---------------------> 0 0 255 1579544448.912749
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 0 600 255 1579544476.448768
TS_READ_RAW----> x = 0, y = 600, pressure = 255
RAW---------------------> 0 0 255 1579544476.464759
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 34 600 255 1579544476.480765
TS_READ_RAW----> x = 34, y = 600, pressure = 255
RAW---------------------> 34 600 255 1579544476.480779
TS_READ_RAW----> x = 34, y = 600, pressure = 255
RAW---------------------> 0 0 0 1579544476.496765
TS_READ_RAW----> x = 0, y = 0, pressure = 0
Took 11 samples...
Bot left : X = 0 Y = 600
RAW---------------------> 0 0 255 1579544476.512755
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 34 600 255 1579544476.528762
TS_READ_RAW----> x = 34, y = 600, pressure = 255
RAW---------------------> 0 600 255 1579544476.544759
TS_READ_RAW----> x = 0, y = 600, pressure = 255
RAW---------------------> 0 0 255 1579544476.560749
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 34 600 255 1579544476.576759
TS_READ_RAW----> x = 34, y = 600, pressure = 255
RAW---------------------> 0 600 255 1579544476.592752
TS_READ_RAW----> x = 0, y = 600, pressure = 255
RAW---------------------> 0 0 255 1579544476.608755
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 0 600 255 1579544483.760769
TS_READ_RAW----> x = 0, y = 600, pressure = 255
RAW---------------------> 0 0 255 1579544483.776760
TS_READ_RAW----> x = 0, y = 0, pressure = 255
RAW---------------------> 519 600 255 1579544483.792764
TS_READ_RAW----> x = 519, y = 600, pressure = 255
RAW---------------------> 519 600 255 1579544483.792776
TS_READ_RAW----> x = 519, y = 600, pressure = 255
RAW---------------------> 0 0 0 1579544483.808764
TS_READ_RAW----> x = 0, y = 0, pressure = 0
Took 11 samples...
Center : X = 0 Y = 600
ts_calibrate: determinant is too small -- 0.000000
Calibration failed. 
```

tparvais, 2020-01-30:
Any idea?

tparvais, 2020-01-30:
Any idea?

HosseinKasaei, 2024-08-05:
4 years later...did you make it to work somehow?

HosseinKasaei, 2024-08-06:
I've found a solution for my case, might help anyone stuck on Waveshare touchscreens working with tslib
#187