libts/tslib #167: 2 version: ts_calibrate: "ts_setup: No such file or directory"

urus-vp, 2020-01-27:
new one... https://github.com/libts/tslib/issues/144
=================================
Hello!
I have the same issue... I just tried to implement your recomendation above, but w/o success...
please, take a look at the attached screenshots.
Do you have any ideas ?
note:
there are three input TouchController devices... but I can get events from event2 only

[image]
[image]

merge, 2020-01-27:
what version of tslib do you run? if you build tslib yourself, you can --enable-debug to see some more details when running ts_calibrate here.

urus-vp, 2020-01-27:
right now I can tell:
1) tslib version: 1.21
2) I use following cmake cmd:
cmake ./configure --enable-static --disable-shared --enable-input=static --enable-linear=static --enable-iir=static --enable-invert=static --enable-median=static --enable-pthres=static --enable-dejitter=static --enable-debounce=static --enable-lowpassskip=static --enable-evthres=static --enable-variance=static ../

urus-vp, 2020-01-27:
I just rebuild tslib on Linux Mint 18.3 Sylvia.
Then copy to target device PDX-057 with Tiny Core & Touch Screen executable files only...
so, now I see following problems: 
[image]

merge, 2020-01-28:
well, so you want to use the median filter but it doesn't seems to be installed so that a program can find it. the plugin libraries ned to be in, for  example, /usr/lib/ts/median.so

urus-vp, 2020-01-28:
mmm, I understand... but I make tslib with STATIC option... and median module too, cmake call: "...--enable-median=static...", so, I guess that all obj files into the tslib modules.
or there are another functionality ?
================================
cmake ./configure --enable-static --disable-shared --enable-input=static --enable-linear=static --enable-iir=static --enable-invert=static --enable-median=static --enable-pthres=static --enable-dejitter=static --enable-debounce=static --enable-lowpass=static --enable-skip=static --enable-evthres=static --enable-variance=static --enable-waveshare=static --enable-touchkit=static ../

xunshichentuo, 2022-10-14:
@urus-vp Have you solved this problem? Thanks.