libts/tslib #168: "ts_open: No such file or directory" message.

thebestofall007, 2020-03-08:
Hello, when I was compiling the latest tslib on my raspberry pi under raspbian for my waveshare 5" screen per the instructions here, [https://github.com/saper-2/rpi-5inch-hdmi-touchscreen-driver](url) I couldn't get tslib to work and get  "ts_open: No such file or directory" when running `sudo TSLIB_CONFFILE=/etc/ts.conf TSLIB_FBDEVICE=/dev/fb0 ts_test`. I have had this problem in raspbian jessie, stretch and buster. 

What went wrong? Is there anything missing?

merge, 2020-03-08:
tslib tries to detect the touchscreen device but seems to fail for your device. it's TSLIB_TSDEVICE you need to set to your /dev/input/eventX device.

Am 8. März 2020 07:24:13 MEZ schrieb thebestofall007 <notifications@github.com>:
> Hello, when I was compiling the latest tslib on my raspberry pi under
> raspbian for my waveshare 5" screen per the instructions here, [...]

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

thebestofall007, 2020-03-08:
Thank you. That did it.

merge, 2020-04-03:
I assume we can close this. thanks.

closed 2020-04-03 by merge