libts/tslib #176: ts_test works, ts_test_mt don't

JonnyMe, 2020-05-04:
Hi!

I'm working on a driver for my device's touchscreen and I'm encountering a strange behaviour..If I run `ts_calibrate` and `ts_test` I can perform all the calibrations and tests but if I run `ts_test_mt` it seems not recording the touches.

Furthermore I noticed that if I run `ts_test_mt` without having touched the screen before I got this behaviour: assuming I touched the device once and then released, if I press again with one finger I don't get the touch but if I press with another finger (keeping the other pressed) it gets the second touch and so on incrementally.

Here is an evtest log for a single tap:
```
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:	pm8941_pwrkey
/dev/input/event1:	himax-touchscreen
Select the device event number [0-1]: 1
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "himax-touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 158 (KEY_BACK)
    Event code 172 (KEY_HOMEPAGE)
    Event code 217 (KEY_SEARCH)
    Event code 330 (BTN_TOUCH)
    Event code 580 (KEY_APPSELECT)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    359
      Min        0
      Max      719
    Event code 1 (ABS_Y)
      Value    644
      Min        0
      Max     1279
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max      200
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        4
    Event code 48 (ABS_MT_TOUCH_MAJOR)
      Value      0
      Min        0
      Max      200
    Event code 50 (ABS_MT_WIDTH_MAJOR)
      Value      0
      Min        0
      Max      200
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max      719
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     1279
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
    Event code 58 (ABS_MT_PRESSURE)
      Value      0
      Min        0
      Max      200
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
Event: time 1586629821.166108, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 320
[... 234 lines cut ...]
TEST ts_read_mt (blocking) 1       ......   PASS
TEST ts_read_mt (blocking) 5       ......   PASS
TEST ts_read_mt (nonblocking) 1    ......   PASS
TEST ts_read_mt (nonblocking) 5    ......   PASS
TEST ts_read    (blocking) 1       ......   PASS
TEST ts_read    (blocking) 5       ......   PASS
TEST ts_read    (nonblocking) 1    ......   PASS
TEST ts_read_raw_mt (blocking) 1   ......   PASS
TEST ts_read_raw_mt (blocking) 5   ......   PASS
TEST ts_read_raw_mt (nonblocking) 1......   PASS
TEST ts_read_raw_mt (nonblocking) 5......   PASS
TEST ts_read_raw(blocking) 1       ......   PASS
TEST ts_read_raw(blocking) 5       ......   PASS
TEST ts_read_raw(nonblocking) 1    ......   PASS
TEST ts_reconfig (1)               ......   PASS
TEST ts_load_module (4)            ......   PASS
======================================================
skip
===================== test run 1 =====================
TEST ts_read_mt (blocking) 1       ......   PASS
TEST ts_read_mt (blocking) 5       ......   FAIL
```
And can't continue...What could be the possible cause for this behaviour? If it could be helpful here is the C source for the driver https://github.com/JonnyMe/linux/blob/49a1bfe5661510b06368c2a148a3feb7eda989a0/drivers/input/touchscreen/himax_852xES.c . 

Thank you!

zedrdave, 2020-05-04:
FWIW: I am encountering the same problem (`ts_test_mt` doesn't detect taps properly: buttons highlight, but do not get triggered, as if it detected the 'down', but not the 'up' event).
I am using a Waveshare 10'1 and tried both raw and waveshare inputs:
```
#module_raw input
module_raw waveshare vid_pid=0483:5710 len=25
```

JonnyMe, 2020-05-04:
I's sorry I messed up with some event report. Now it seems working, I will close soon after some testing,

Thank you!

JonnyMe, 2020-05-04:
Closing, for the records I was not sending correctly `ABS_MT_TOUCH_MAJOR` and `ABS_MT_WIDTH_MAJOR`...also I was firing `ABS_MT_PRESSURE` with a fixed value (removed `ABS_MT_PRESSURE` part as device does not record pressure),

Thank you again

closed 2020-05-04 by JonnyMe