libts/tslib #187: Raspberry 4 & Yocto & Xorg : tslib is flooding my console with "tslib: Selected device is not a Linux input event device" error

Julien-Faure, 2021-04-30:
[     7.761]
X.Org X Server 1.20.8
X Protocol Version 11, Revision 0
[     7.762] Build Operating System: Linux
[     7.762] Current Operating System: Linux raspberrypi4-64 5.4.72-v8 #1 SMP PREEMPT Mon Oct 19 11:12:20 UTC 2020 aarch64
[     7.762] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 video=HDMI-A-1:800x480M@59 smsc95xx.macaddr=DC:A6:32:94:BA:C9 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait logo.nologo vt.global_cursor_default=0 psplash=false
[     7.762] Build Date: 29 March 2020  08:21:46PM
[     7.762]
[     7.762] Current version of pixman: 0.38.4
[     7.762]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[     7.762] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[     7.762] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Mar  9 12:35:19 2018
[     7.786] (==) Using config file: "/etc/X11/xorg.conf"
[     7.786] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[     7.808] (==) No Layout section.  Using the first Screen section.
[     7.808] (==) No screen section available. Using defaults.
[     7.808] (**) |-->Screen "Default Screen Section" (0)
[     7.808] (**) |   |-->Monitor "<default monitor>"
[     7.809] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[     7.809] (==) Automatically adding devices
[     7.809] (==) Automatically enabling devices
[     7.809] (==) Automatically adding GPU devices
[     7.809] (==) Max clients allowed: 256, resource mask: 0x1fffff
[     7.816] (WW) The directory "/usr/share/fonts/X11/misc/" does not exist.
[     7.816]    Entry deleted from font path.
[     7.816] (WW) The directory "/usr/share/fonts/X11/TTF/" does not exist.
[     7.816]    Entry deleted from font path.
[     7.816] (WW) The directory "/usr/share/fonts/X11/OTF/" does not exist.
[     7.816]    Entry deleted from font path.
[     7.816] (WW) The directory "/usr/share/fonts/X11/Type1/" does not exist.
[     7.816]    Entry deleted from font path.
[     7.816] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[     7.816]    Entry deleted from font path.
[     7.816] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[     7.816]    Entry deleted from font path.
[     7.816] (==) FontPath set to:

[     7.816] (==) ModulePath set to "/usr/lib/xorg/modules"
[     7.816] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[     7.816] (II) Loader magic: 0x557f500c28
[     7.816] (II) Module ABI versions:
[     7.816]    X.Org ANSI C Emulation: 0.4
[     7.816]    X.Org Video Driver: 24.1
[     7.816]    X.Org XInput driver : 24.1
[     7.816]    X.Org Server Extension : 10.0
[     7.817] (II) xfree86: Adding drm device (/dev/dri/card1)
[     7.828] (II) xfree86: Adding drm device (/dev/dri/card0)
[     7.829] (II) no primary bus or device found
[     7.829]    falling back to /sys/devices/platform/gpu/drm/card1
[     7.829] (II) LoadModule: "glx"
[     7.853] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[     8.117] (II) Module glx: vendor="X.Org Foundation"
[     8.117]    compiled for 1.20.8, module version = 1.0.0
[     8.117]    ABI class: X.Org Server Extension, version 10.0
[     8.117] (==) Matched modesetting as autoconfigured driver 0
[     8.117] (==) Matched fbdev as autoconfigured driver 1
[     8.117] (==) Assigned the driver to the xf86ConfigLayout
[     8.117] (II) LoadModule: "modesetting"
[     8.118] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[     8.127] (II) Module modesetting: vendor="X.Org Foundation"
[     8.127]    compiled for 1.20.8, module version = 1.20.8
[     8.127]    Module class: X.Org Video Driver
[     8.127]    ABI class: X.Org Video Driver, version 24.1
[     8.127] (II) LoadModule: "fbdev"
[     8.128] (WW) Warning, couldn't open module fbdev
[... 79 lines cut ...]
[     9.361]    Module class: X.Org XInput Driver
[     9.361]    ABI class: X.Org XInput driver, version 24.1
[     9.361] (II) Using input driver 'tslib' for 'raspberrypi-ts'
[     9.361] (**) raspberrypi-ts: always reports core events
[     9.361] (**) raspberrypi-ts: always reports core events
[     9.368] (**) Option "config_info" "udev:/sys/devices/platform/soc/soc:firmware/soc:firmware:touchscreen/input/input0/event0"
[     9.368] (II) XINPUT: Adding extended input device "raspberrypi-ts" (type: TOUCHSCREEN, id 6)
[     9.368] (**) raspberrypi-ts: (accel) keeping acceleration scheme 1
[     9.368] (**) raspberrypi-ts: (accel) acceleration profile 0
[     9.368] (**) raspberrypi-ts: (accel) acceleration factor: 2.000
[     9.368] (**) raspberrypi-ts: (accel) acceleration threshold: 4
[     9.369] (II) config/udev: Adding input device raspberrypi-ts (/dev/input/mouse0)
[     9.369] (**) raspberrypi-ts: Applying InputClass "tslib touchscreen"
[     9.369] (II) Using input driver 'tslib' for 'raspberrypi-ts'
[     9.369] (**) raspberrypi-ts: always reports core events
[     9.369] (**) raspberrypi-ts: always reports core events
[     9.369] (**) Option "config_info" "udev:/sys/devices/platform/soc/soc:firmware/soc:firmware:touchscreen/input/input0/mouse0"
[     9.369] (II) XINPUT: Adding extended input device "raspberrypi-ts" (type: TOUCHSCREEN, id 7)
[     9.369] (**) raspberrypi-ts: (accel) keeping acceleration scheme 1
[     9.369] (**) raspberrypi-ts: (accel) acceleration profile 0
[     9.369] (**) raspberrypi-ts: (accel) acceleration factor: 2.000
[     9.369] (**) raspberrypi-ts: (accel) acceleration threshold: 4
[    24.290] (II) modeset(0): Disabling kernel dirty updates, not required.

```

Julien-Faure, 2021-05-03:
I don't find the problem but I found an alternative :

I changed my DM, from LXDE to XFCE4.

HosseinKasaei, 2024-08-06:
> I don't find the problem but I found an alternative :
> 
> I changed my DM, from LXDE to XFCE4.

I have exactly the same problem but I'm not using any desktop environment just openbox as a window manager. Do you have any idea how I can fix that?