libts/tslib #188: why tslib not working in pyqt5 application?

sanluku, 2021-08-06:
I can calibrate my touch screen with tslib, and generate pointercal.
And i export like this:
export TSLIB_TSDEVICE=/dev/input/event4
export TSLIB_CONFFILE=/etc/ts.conf
export TSLIB_CALIBFILE=/etc/pointercal

export QT_QPA_EGLFS_TSLIB=1
export QT_QPA_FB_TSLIB=1
export QT_QPA_EVDEV_TOUCHSCREEN_PARAMETERS=/dev/input/event4:rotate=0
export QT_QPA_PLATFORM_PLUGIN_PATH=/usr/lib/plugins
export QT_QPA_PLATFORM=linuxfb:tty=/dev/fb0
export QT_QPA_GENERIC_PLUGINS=tslib:/dev/input/event4

but in my application, it's still not calibrate, it seems calibrate file didn't work.
Did i do something wrong?

merge, 2021-08-10:
how does your /etc/ts.conf file look like?

sanluku, 2021-08-10:
I didn't make any change.
It's look like:

module_raw input
module pthres pmin=1
module dejitter delta=100
module linear