libts/tslib #192: ts_uinput - open /dev/uinput: No such file or directory

OOHehir, 2021-12-03:
Hello,

I'm running v1.22 (built using Yocto) on a SAMA5D4 Xplained Demo board. 

ts_calibrate & ts_test work as expected but when it comes to ts_uinput I get:

open /dev/uinput: No such file or directory

My /dev/input/

`
drwxr-xr-x 2 root root      80 Jan  1  2012 by-path
crw-rw---- 1 root input 13, 64 Jan  1  2012 event0
crw-rw---- 1 root input 13, 65 Jan  1  2012 event1
lrwxrwxrwx 1 root root       6 Jan  1  2012 touchscreen0 -> event0

`
I've tried:

TSLIB_TSDEVICE=/dev/input/event0

& then 

./ts_uinput -d -v

But the same result.

My ts.conf:
`# Uncomment if you wish to use the linux input layer event interface
module_raw input

# Uncomment if you're using a Sharp Zaurus SL-5500/SL-5000d
# module_raw collie

# Uncomment if you're using a Sharp Zaurus SL-C700/C750/C760/C860
# module_raw corgi

# Uncomment if you're using a device with a UCB1200/1300/1400 TS interface
# module_raw ucb1x00

# Uncomment if you're using an HP iPaq h3600 or similar
# module_raw h3600

# Uncomment if you're using a Hitachi Webpad
# module_raw mk712

# Uncomment if you're using an IBM Arctic II
# module_raw arctic2

module pthres pmin=1
# module variance delta=30  <-- commented out after reading previous issue report but no change
module dejitter delta=100
module linear`

uname -a returns:

Linux sama5d4-xplained-sd 5.10.50-linux4sam-2021.04 #1 Thu Jul 29 14:58:03 UTC 2021 armv7l armv7l armv7l GNU/Linux

playya, 2021-12-03:
If you want to Feed Touchscreen Events into evdev using uinput you need to
enable CONFIG_UINPUT in your Kernel config:
https://elixir.bootlin.com/linux/latest/source/drivers/input/misc/Kconfig#L513

If this ist already enabled as a Module you need to load the Module e.g.
using modprobe: modprobe uinput

OOHehir, 2021-12-03:
Many thanks for the prompt reply!

The module was not built in my version of Yocto/ Poky so have to do a rebuild..

I'll report back when I have a result.

For anyone else looking at this I'm trying for the altering the build:

source oe-init-build-env build-microchip

bitbake -c menuconfig virtual/kernel

Modifying 'User level driver support' to built-in (*)

playya, 2021-12-03:
Pleased Close this Ticket as invalid because it's Not an tslib Bug.

If you have questions regarding yocto pleased contact the yocto mailing list

OOHehir, 2021-12-05:
Hello,
Yes, this has solved the issue:

/dev/input/ now has:

```
drwxr-xr-x  2 root root      80 Jan  1  2012 by-path
crw-rw----  1 root input 13, 64 Jan  1  2012 event0
crw-rw----  1 root input 13, 65 Jan  1  2012 event1
crw-rw----  1 root input 13, 66 Mar  9 12:59 event2
lrwxrwxrwx  1 root root       6 Mar  9 12:59 touchscreen0 -> event2
```

Thanks for the help!

closed 2021-12-05 by OOHehir