libts/tslib #193: how to change the touch area

19504643, 2021-12-21:
I'm using a 4.3 inch 800*480 LCD and 12 bit touch chip, now only a very small area can get the cursor to move, and  the ts_calibration not working, how to change the touch area to the full screen
```
./ts_calibrate
xres = 800, yres = 480
```
when I run ts_test, touch the upper left corner and lower right corner, it returns minimum and maximum value, but the tslib only get the value from 0 to 800 in horizontal and 0 to 480 in vertical.
```
./ts_test
3867.380129:    220    379    255
3867.400118:    217    374    255
3867.420121:    215    389    255
3870.300119:   3892   3759    255
3870.310131:   3898   3765    255
3870.330114:   3903   3768    255
```
here is the conf:
```
cat ../etc/ts.conf
# Access plugins
################

# Uncomment if you wish to use the linux input layer event interface
module_raw input

# For other driver modules, see the ts.conf man page

# Filter plugins
################

# Uncomment if first or last samples are unreliable
# module skip nhead=1 ntail=1

# Uncomment if needed for devices that measure pressure
module pthres pmin=1

# Uncomment if needed
# module debounce drop_threshold=40

# Uncomment if needed to filter spikes
# module median depth=5

# Uncomment to enable smoothing of fraction N/D
# module iir N=6 D=10

# Uncomment if needed
# module lowpass factor=0.1 threshold=1

# Uncomment if needed to filter noise samples
module dejitter delta=100

# Uncomment to define threshold in number of events from device
# module evthres N=5

# Uncomment and adjust if you need to invert an axis or both
# module invert x0=800 y0=480

# Uncomment to use ts_calibrate's settings
module linear

```
[image]

iqbal-em, 2022-09-16:
Hello, How are you ? I have same issue like you. Have you already solve the problem?

Thanks

yifanshu02, 2023-03-09:
hello，how r u? i have same issue like you.Have you already solve the problem??

yifanshu02, 2023-03-09:
i think the ts_calibrate is failed...

yifanshu02, 2023-03-09:
solved！！
```
# Uncomment to use ts_calibrate's settings
#module linear
```