libts/tslib #209: autogen.sh fails on older system

bigguiness, 2023-12-05:
I'm trying to add a touchscreen to an old BeagleBone system running wheezy (Debian 7.11). I know it's old, I know I should update it, but it works...

When I try to use the touchscreen I get a message that says:
selected device is not a touchscreen I understand

I tracked that message down to coming from tslib so I thought I would try updating it.

I cloned the repo, but when I try to run autogen.sh I get this:

$ ./autogen.sh
configure.ac:64: error: possibly undefined macro: AC_DISABLE_STATIC
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:65: error: possibly undefined macro: AC_ENABLE_SHARED
configure.ac:66: error: possibly undefined macro: AC_LIBTOOL_DLOPEN
configure.ac:67: error: possibly undefined macro: AC_PROG_LIBTOOL
autoreconf: /usr/bin/autoconf failed with exit status: 1

Is there any chance of getting this to work?

bigguiness, 2023-12-05:
Appears I was missing libtool. Installed that and the autogen.sh worked.

I then did:
$ ./configure
$ make
$ sudo make install

But when i try to run ts_test, just to see if everything works, I get this:

$ sudo ts_test
ts_test: error while loading shared libraries: libts.so.0: cannot open shared object file: No such file or directory

Any ideas? Did I miss something?

merge, 2024-02-19:
look for the installed libts.so.0*. where is it? what other libraries are there?

btw. you can always use release-tarballs. no need to run autoreconf on those.