libts/tslib #227: Wrong input in PyQt6 with tslib

defomans, 2024-10-14:
Hello. I run compiled tslib on RPi zero 2w. I have an ili9341 2.8inch display with xpt2046 touch. I initialize environment with these parameters:
export LD_LIBRARY_PATH=/usr/local/lib
export TSLIB_CONSOLEDEVICE=none
export TSLIB_FBDEVICE=/dev/fb1
export TSLIB_TSDEVICE=/dev/input/event0
export TSLIB_CALIBFILE=/home/pi/pointercal
export TSLIB_CONFFILE=/usr/local/etc/ts.conf
export TSLIB_PLUGINDIR=/usr/local/lib/ts
ts_calibrate
I run the ts_test and I get the perfect input.
Then I run ts_uinput -d -v and it generates /dev/input/event1
Then I export QT_QPA_EVDEV_TOUCHSCREEN_PARAMETERS=/dev/input/event1:rotate=0

I run the PyQt6 app on linuxfb. I can see that now I can get the touch input more or less working on my display in the PyQT6 app, but it's definitely far from accurate. I cannot click on the items in the bottom of the QComboBox unless I select the middle item and drag down. When I click on items I often get weird behaviour like double clicking. For some reason, I can open combobox by clicking on the place where it's not even there. I also show the virtual keyboard and when I click on one button the totally different one gets clicked. So, I can definitely see that the input is off.
So, what could be a problem and how to solve it? I literally searched everywhere and cannot find a solution. Thank you.

merge, 2024-10-17:
what's your ts.conf. are you sure you use the same environment setting for ts_test and qt (ts_uinput)?

defomans, 2024-10-17:
**Here are the contents of ts.conf located in /usr/local/etc/ts.conf**
# Access plugins
################
# Uncomment if you wish to use the linux input layer event interface
module_raw input
# For other driver modules, see the ts.conf man page
# Filter plugins
################
# Uncomment if first or last samples are unreliable
# module skip nhead=1 ntail=1
# Uncomment if needed for devices that measure pressure
module pthres pmin=1
# Uncomment if needed
module debounce drop_threshold=40
# Uncomment if needed to filter spikes
module median depth=5
# Uncomment to enable smoothing of fraction N/D
# module iir N=6 D=10
# Uncomment if needed
# module lowpass factor=0.1 threshold=1
# Uncomment if needed to filter noise samples
module dejitter delta=100
# Uncomment to define threshold in number of events from device
# module evthres N=5
# Uncomment and adjust if you need to invert an axis or both
# module invert x0=800 y0=480
# Uncomment to use ts_calibrate's settings
module linear
# Uncomment to drop events outside of the framebuffer
# module crop

I'm not sure I use the same environment for ts_test and ts_uinput. But I initialize the environment as described before, then run ts_uinput -d -v and in the same terminal window run pyqt app (all these steps are done within the same terminal window). I expected that there's only 1 environment used for all of these, when I calibrate the display using ts_calibrate. Could you please elaborate further on this? What should I do to use the same environment and finally get my input right from ts_uinput?

defomans, 2024-10-17:
I tried to run all the commands to set up the environment before everything else within my pyqt6 app:
```
os.environ["QT_QPA_PLATFORM"] = "linuxfb:fb=/dev/fb1"
os.environ["LD_LIBRARY_PATH"] = "/usr/local/lib"
os.environ["TSLIB_CONSOLEDEVICE"] = "none"
os.environ["TSLIB_FBDEVICE"] = "/dev/fb1"
os.environ["TSLIB_TSDEVICE"] = "/dev/input/event2"  # event2 is my physical touchscreen device
os.environ["TSLIB_CALIBFILE"] = "/home/pi/pointercal"
os.environ["TSLIB_CONFFILE"] = "/usr/local/etc/ts.conf"  # also tried /etc/ts.conf - no effect  
os.environ["TSLIB_PLUGINDIR"] = "/usr/local/lib/ts"
subprocess.call(['sudo', 'chmod', '+0666', '/dev/uinput'])  # without this command I get permission denied from ts_uinput
result = subprocess.run(['ts_uinput', '-d', '-v']) # Generates /dev/input/event3
os.environ["QT_QPA_EVDEV_TOUCHSCREEN_PARAMETERS"] = "/dev/input/event3:rotate=0"
```
And nothing really changes, moreover the input got even more off. Now I cannot even open ComboBox when clicking on it, only by clicking on random locations where it's not there.

defomans, 2024-10-18:
Interesting thing is that even if I change rotation in `export QT_QPA_EVDEV_TOUCHSCREEN_PARAMETERS=/dev/input/event1:rotate=0`  to 90 or any other angle, It doesn't change anything, the input is the same, still wrong. What could be the problem?

defomans, 2024-10-19:
I installed pyqt5 to try if pyqt6.4.3 had problems, but it's all the same, doesn't work either. I reinstalled tslib multiple times with different ways, didn't help. I literally tried all the environment variables for tslib and pyqt I could find for the internet. Nothing works. What I found is that touch input is okay, it's just reversed 90 degrees compared to my current screen orientation. And I cannot rotate this touch at all, nothing helps, idk why. It seems like pyqt5 or 6 doesn't recognize tslib at all, because no matter what I do, even if I don't initialize tslib environment, I get the same input and the same problem in pyqt. What can be wrong?